ros2/demos #11: pendulum_control: fails to build

vmayoral, 2015-08-24:
@jacquelinekay this package doesn't build:
```
-- Build files have been written to: /home/victor/ros2_ws/build/pendulum_control
==> '. /home/victor/ros2_ws/build/pendulum_control/cmake__build.sh && /usr/bin/make -j3 -l3' in '/home/victor/ros2_ws/build/pendulum_control'
[ 50%] [100%] Building CXX object CMakeFiles/pendulum_demo__rmw_opensplice_cpp.dir/src/pendulum_demo.cpp.o
Building CXX object CMakeFiles/pendulum_demo.dir/src/pendulum_demo.cpp.o
Linking CXX executable pendulum_demo
Linking CXX executable pendulum_demo__rmw_opensplice_cpp
/usr/bin/ld: CMakeFiles/pendulum_demo__rmw_opensplice_cpp.dir/src/pendulum_demo.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [pendulum_demo__rmw_opensplice_cpp] Error 1
make[1]: *** [CMakeFiles/pendulum_demo__rmw_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/pendulum_demo.dir/src/pendulum_demo.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [pendulum_demo] Error 1
make[1]: *** [CMakeFiles/pendulum_demo.dir/all] Error 2
make: *** [all] Error 2

```
Tried adding `-lpthread ` https://github.com/erlerobot/demos/commit/ea77763e595513b256b2075f9afa5428e906339a but doesn't seem to help. Any ideas?

jacquelinekay, 2015-08-24:
Hmm.

Did you clone https://github.com/ros2/rttest into your workspace? That might provide the missing symbols for `pthread`.

vmayoral, 2015-08-24:
rttest is there yeap. Bumped with it while following https://github.com/ros2/examples/wiki (did it in a fresh trusty chroot).

jacquelinekay, 2015-08-24:
Can you try again with branch `fix_cmake` of rttest?

https://github.com/ros2/rttest/tree/fix_cmake

my current theory is that the package relies on rmw_connext to find pthread for us:

https://github.com/ros2/rmw_connext/blob/master/connext_cmake_module/cmake/Modules/FindConnext.cmake#L255

which is why we haven't seen the issue yet on Jenkins or in my testing, because I always build with Connext.

jacquelinekay, 2015-08-24:
@codebot also saw this issue, so it is not an isolated incident.

dirk-thomas, 2015-08-24:
Should be fixed with ros2/rttest#4

closed 2015-08-24 by dirk-thomas