ros2/demos #61: cam2image__rmw_connext_cpp fails

jacquelinekay, 2016-02-18:
Running showimage in another terminal, if I start cam2image I get the error:

```
Publishing image #1
COMMENDSrWriterService_write:!write. Reliable large data requires asynchronous writer.
PRESPsWriter_writeInternal:!srw->write
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to publish message: failed to publish message, at /home/jackie/code/ros2_ws/src/ros2/rmw_connext/rmw_connext_cpp/src/functions.cpp:410
Aborted (core dumped)
```

jacquelinekay, 2016-02-18:
As you might expect, running the demo with `-r 0` causes it to work

mikaelarguedas, 2017-02-01:
dunno when that was fixed but it was not reported in recent release testing and cant reproduce locally. 
If anybody can confirm we can close this and update tutorial page accordingly

dhood, 2017-04-26:
closing this because tests for these executables were added in https://github.com/ros2/demos/pull/83

closed 2017-04-26 by dhood