ros2/demos #68: demos should print reasonable error message when no camera is present

dirk-thomas, 2016-06-01:
Currently `camera_node` and `image_pipeline_*` just crash without any notice. `cam2image` e.g. prints an error message (`Could not open video stream`) which is the desired behavior.

wjwwood, 2016-06-01:
We can add error handling to all of the demos (`try {...} catch(...) {...}`) or we could try to put something automatic together with some combination of these c++11 functions:

- http://en.cppreference.com/w/cpp/error/set_terminate
- http://en.cppreference.com/w/cpp/error/current_exception
- http://en.cppreference.com/w/cpp/error/uncaught_exception

jacquelinekay, 2016-06-01:
I would guess from this that Windows doesn't print the message for an exception when it is thrown by default? That's annoying.

+1 for checking for uncaught_exception in the destructor of CameraNode, rather than wrapping the instantiation of CameraNode with try/catch everywhere.

dirk-thomas, 2016-06-01:
@jacquelinekay The exception is raised in https://github.com/ros2/demos/blob/82bbf8fefafcfed5f9bf8e844821ab81b564f1bf/intra_process_demo/src/image_pipeline/camera_node.hpp#L43 which makes the object creation fail. There is no invocation of the destructor anywhere.

wjwwood, 2016-06-01:
We could have a standard "termination_handler" that uses "current_exception" to print a useful message, and then we'd only need to install it with "set_terminate" in each demo. That's slightly less intrusive than wrapping all mains with a try-catch, but also slightly more magical.

jacquelinekay, 2016-06-01:
hmm yeah, I was misinterpreting the example on http://en.cppreference.com/w/cpp/error/uncaught_exception

Karsten1987, 2016-10-04:
any news on that? I believe the actual error comes from https://github.com/ros2/demos/blob/master/intra_process_demo/src/image_pipeline/camera_node.hpp#L39 where it blindly opens device 0

Karsten1987, 2016-10-04:
```
c:\dev\ci-ros2-fastrtps\ros2-windows>camera_node
Could not open video stream! Exiting..
```

fixed in https://github.com/ros2/demos/pull/81

closed 2016-10-04 by Karsten1987