ros2/demos #69: strange overwritten error with

wjwwood, 2016-06-01:
For example on my machine, while testing alpha 6:

```
% two_node_pipeline__rmw_fastrtps_cpp
[rmw|error_handling.c:93] error string being overwritten: not implemented, at /Users/william/ros2_ws/src/eProsima/ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp/src/functions.cpp:1768
Published message with value: 0, and address: 0x7FFA898004B0
 Received message with value: 0, and address: 0x7FFA898004B0
Published message with value: 1, and address: 0x7FFA885AE6D0
 Received message with value: 1, and address: 0x7FFA885AE6D0
Published message with value: 2, and address: 0x7FFA885AE6D0
 Received message with value: 2, and address: 0x7FFA885AE6D0
Published message with value: 3, and address: 0x7FFA885AE6D0
 Received message with value: 3, and address: 0x7FFA885AE6D0
^Csignal_handler(2)
```

It is because of this nonsense that I had in the fastrtps function to get the graph guard condition:

https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L1768-L1769

I'm working on a fix.

wjwwood, 2016-06-01:
Fix: https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/32

jacquelinekay, 2016-06-01:
Your fix got merged, so we can close this, right?

wjwwood, 2016-06-01:
Yup.

closed 2016-06-01 by wjwwood