ros2/demos #79: showimage demo fails with OpenSplice

dirk-thomas, 2016-09-16:
On Linux and OS X the test fails:
* http://ci.ros2.org/job/ci_linux/lastCompletedBuild/testReport/
* http://ci.ros2.org/job/ci_osx/lastCompletedBuild/testReport/

On Windows it fails to compile:
* http://ci.ros2.org/job/ci_windows/1659/console

mikaelarguedas, 2017-09-08:
This test passed on [Windows](http://ci.ros2.org/job/ci_windows_opensplice67/8) and [Linux](http://ci.ros2.org/job/ci_linux/3193) on the latest jobs. Keeping it open until MacOS test pass as well

@dirk-thomas FYI

dirk-thomas, 2017-09-12:
I will close this for now since the problem seems to be fixed on all the platforms we can currently test. If it happens to fail on OS X in the future we can reopen this.

closed 2017-09-12 by dirk-thomas