ros2/demos #85: Quality of service demos in python

dhood, 2016-11-01:
**Context:**
The [image_tools demo](https://github.com/ros2/demos/tree/master/image_tools) is used in the [“Use quality-of-service settings to handle lossy networks”](https://github.com/ros2/ros2/wiki/Quality-Of-Service) tutorial. This demo currently only exists for C++. We have since added a [python client library for ROS 2](https://github.com/ros2/ros2/wiki/Python-Programming).

**To do:**
- [ ] Replicate the image_tools demo using the python client library.

- [ ] Incorporate it into the [quality of service tutorial](https://github.com/ros2/ros2/wiki/Quality-Of-Service).

**Pointers:**
- This PR has examples of using different QoS profiles in python nodes: https://github.com/ros2/examples/pull/132/files
- This OpenCV tutorial shows how to show images in python: http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_gui/py_image_display/py_image_display.html
- This C++ image_tools demo shows how to list a dependency on OpenCV: https://github.com/ros2/demos/blob/16c3bc5d41462bbace90bff66d45ad1f3e42f91a/image_tools/package.xml#L11

kgreenek, 2016-11-23:
Was this closed by:
https://github.com/ros2/examples/pull/132
?

dhood, 2016-11-28:
Not quite, but they are indeed related. The examples in ros2/examples#132 (e.g. `qos_listener_py.py`) are quite simple - they just send/receive strings. The demo being referred to here is with images, because it's easier to "feel" the latency in the [quality of service demo](https://github.com/ros2/ros2/wiki/Quality-Of-Service) with a lagging webcam stream than it is to with strings.

clalancette, 2017-01-06:
I'm going to take a crack at doing this to get my feet wet.

mikaelarguedas, 2018-02-22:
As this is low priority and blocked by the availability of OpenCV3 python3 bindings (ubuntu bionic), assigning it to the untargeted milestone