ros2/demos #100: [lifecycle] installed python files are not executable

wjwwood, 2016-12-19:
They are in `bin`, but not executable and so `which` doesn't find them:

```
% which lifecycle_listener
/tmp/testing_beta1/ros2-osx/bin/lifecycle_listener
% which lifecycle_demo_launch.py
lifecycle_demo_launch.py not found
```

However, this is how the user is recommended to execute them:

> ```
> python3 `which lifecycle_service_client_py.py`
> ```
>
> -- https://github.com/ros2/ros2/wiki/Managed-Nodes

Not sure if this works on Linux as-is, but it doesn't work with `zsh` on OS X (my machine).

dirk-thomas, 2016-12-19:
The file is executable when building from source. But it doesn't have a shebang line. Therefore the instructions worked for me. But this should certainly be fixed.

Karsten1987, 2016-12-19:
Oh, I am sorry. Didn't know that.
So I'll update my python script with:

```
#!/usr/bin/env python
```
Is that correct? Interestingly, all python demos don't have the shebang either. Should that be corrected then as well?

dirk-thomas, 2016-12-19:
Can you point to the specific case you are referring to. At least some are being installed as `console_scripts` from the `setup.py` file which will take care of that.

dirk-thomas, 2016-12-19:
Is anyone working on a fix for this?

dirk-thomas, 2016-12-19:
I created #101. Can one of you please confirm that it fixes the problem for you, too.

PR #101 (2016-12-19): add missing shebang line and fix permissions
Fixes #100.
closed 2016-12-19 by dirk-thomas