ros2/demos #111: two_node_pipeline demo has consumer / producer swapped

mkhansen-intel, 2017-02-08:
I was doing the IPC tutorial and noticed the two_node_pipeline has a minor bug. In the code, it is setting the names of the consumer & producer to the opposite object.
`auto producer = std::make_shared<Producer>("consumer", "number");`
`auto consumer = std::make_shared<Consumer>("producer", "number");`

I'm happy to submit the 2 line PR to fix this if you want. I don't know if you want external PRs at this point.

wjwwood, 2017-02-08:
Lines in question: https://github.com/ros2/demos/blob/803960e4f612bb2b4c1c4232bd5ca0883edee0db/intra_process_demo/src/two_node_pipeline/two_node_pipeline.cpp#L80-L81

Doh, yeah, a pr would be great. Thanks!

mkhansen-intel, 2017-02-09:
Fixed by pr #112

closed 2017-02-09 by mkhansen-intel