ros2/demos #128: Composition examples require network availability to run

edwinferretjp, 2017-05-17:
[I'm not sure if this is a bug or just not implemented yet but the documentation is nonetheless ambiguous.]

**Problem**
The composition examples (https://github.com/ros2/ros2/wiki/Composition) all require network access to run. 

**Expectation**
Even though the documentation suggests that messages being published/subscribed to in the same process will be automatically copied (or a pointer passed depending), the underlying DDS is still used to perform the communication and so network access is required to run the examples.

**Environment**
Tested on MacOS 10.11.6 with ROS2 Beta1 compiled from source. 

**Tests**
Tested the following with/without network access and with/without multi-cast loopback through localhost:
api_composition
manual_composition

**Questions**
Other code which does not yet appear to be documented seems to hint at solving (or at least trying to solve) this issue by specifying a bool use_intra_process_comms flag (e.g. src/ros2/rclcpp/rclcpp/include/rclcpp/create_publisher.hpp). If we inherit all our nodes from LifecycleNode, for example, will the intra-process memory copies(pointer passing) "just work"?

As an aside, which node type is the preferred one to inherit from? Node (as in most of the examples)? LifecycleNode? Something else?

dirk-thomas, 2017-05-17:
The intra-process communication only avoids copying the actual message data (by passing a reference). But the metadata about the message are still passed through DDS (to achieve consistent behavior with other messages as well as when toggling intra-process comms on or off). I don't think there is a public design doc describing this yet (@wjwwood please correct me if I am wrong).

So yes, it is expected that intra-process comms doesn't work if you don't have a local loopback. Depending on the DDS vendor you might be able to use unicast (instead of multicast) but we still require non-intra-process comms to be available.

> As an aside, which node type is the preferred one to inherit from? Node (as in most of the examples)? LifecycleNode? Something else?

Currently there are `Node`, `Component`, and `LifecycleNode`. At the current time we don't have a clear recommendation yet. We haven't used the later two enough to give a solid recommendation.

wjwwood, 2017-05-17:
> I don't think there is a public design doc describing this yet (@wjwwood please correct me if I am wrong).

There is not. The closest thing is our ROSCon 2015 presentation:

http://roscon.ros.org/2015/presentations/state-of-ros2.pdf

Or maybe the documentation of the `IntraProcessManager` in `rclcpp`:

http://docs.ros2.org/beta1/api/rclcpp/classrclcpp_1_1intra__process__manager_1_1_intra_process_manager.html#details

There are lots of nice details in that API documentation, but nothing like an overarching design doc with our design decisions, like to use a DDS message to communicate the metadata of the intra process message to avoid duplicating pub/sub behavior in our own queueing.

edwinferretjp, 2017-05-19:
Many thanks for these quick responses! I must add that ROS2 is a fantastic platform and pretty much everything has worked "as advertised"!

Thank you for the links, and I appreciate this was apparently a conscious design decision, but I still don't really understand why you need to send the meta-data over DDS as well, unless that is the only way that in-process publishers/subscribers can make each other aware that there is actual data available. Is that the take-away from this statement in: http://docs.ros2.org/beta1/api/rclcpp/classrclcpp_1_1intra__process__manager_1_1_intra_process_manager.html#details
_"The subscription passes a publisher id, message sequence pair which uniquely identifies the message instance _it was suppose to receive_ as well as the subscriptions unique id."_

What are the cases where it is necessary to send meta-data over DDS but not the data itself? I guess I'm still missing something.

I know this probably won't be a common usage pattern of ROS2 but, as far as I can tell, by disallowing the no-network/no-loopback option it would be impossible for a ROS2 program where, for example, all nodes run in a single process, to run on, say, an iOS/Android device that is connected to a 3G/4G network or perhaps connected to no network. I appreciate this is probably not a major concern for most developers but I see this as a common use case for us where limited functionality is available within an Android application when it's connected to a telephony network and even more functionality is available(e.g. by interacting with other nodes on the network) when the Android device is connected to a network. However, with the current situation any nodes local to the Android device itself simply won't work with a WiFi network. 

It's a shame because you have a fantastic platform whereby you can launch nodes at compile-, link-, or run-time, arbitrarily distribute nodes across different machines on the same network, that each might be running different OSes, as well as running all nodes on a single machine in the same process and (as far as I can see) almost all possible combinations in-between. You even allow multiple providers for the underlying DDS. This seems to cover everything for embedded-only use as well, except for the case where a machine has no network access or loopback.  

You mentioned that there might be DDS providers that allow unicast instead of multicast but this still requires a network or loopback so I don't think that solves anything. A DDS provider that, like ROS2, allowed messages to be passed in-memory and optionally didn't do discovery and communication over a network would do the trick but that it obviously not the main purpose of DDS so I doubt it's been implemented by anyone.

If you have any other ideas for how this might be achieved for the example of an Android local-process app case (described above) I'd very much like to hear them. Perhaps there's a simple non-ROS2, non-DSS solution that I'm missing. 
Thanks.

dirk-thomas, 2017-05-19:
> I know this probably won't be a common usage pattern of ROS2 but, as far as I can tell, by disallowing the no-network/no-loopback option it would be impossible for a ROS2 program where, for example, [...]

I definitely see the use case of running everything in a single process (with no outside connectivity) as very interesting. Especially when considering to apply that for products which don't need a "debug" port or intentionally don't have one.

I see different options here:
* First: as you mention a DDS vendor might provide a custom transport which doesn't rely on having a network interface at all, e.g. just using shared memory. Some vendors certainly implement such optimizations but I am not sure if that covers all communication or only the discovery. I guess someone could try that to figure out if something like that is already available.
* Second: shouldn't even a device without an external network connection be able to have a local loopback interface which would be sufficient for DDS implementations to discover participants on the same device? If that is the case I don't see a problem running ROS 2 on such a device even if it has no (or only very poor) external connectivity. I am not sure how common / uncommon the situation is that some deviced don't have a loopback interface - maybe someone can clarify that?
* Third: it should be possible to implement the RMW interface with a different backend. It doesn't even have to be DDS. So any other communication framework which doesn't require any network interface would be a candidate for this use case. But that would certainly be some effort :wink:

wjwwood, 2017-05-25:
> What are the cases where it is necessary to send meta-data over DDS but not the data itself? I guess I'm still missing something.

We gain a benefit by passing meta-data over DDS but not the actual data when the data is very large. A big image might go across several UDP packets whereas this lean meta-data is pretty much guaranteed to be one UDP packet. By sending the meta-data over a identically configured topic to the actual data topic, we remove the need to reimplement discovery and QoS settings like history length. It also means that the transaction is observable externally which improves our ability to do introspection.

It was also easier to implement, so we took that route. We knew that in the future we might need to reimplement intra-process using our own internal queues and logic, but we wanted to start here.

It should still be possible in principle to configure a DDS vendor to not use any network, though I haven't tried.

edwinferretjp, 2017-05-26:
Thank you both @dirk-thomas and @wjwwood. All the answers help clarify the design decisions involved. I see now that in the future you could still allow ROS2 to not require using a DDS at all (although you'd have to re-implement the intra-process mechanism) and that leaving it up to the DDS to implement a more efficient solution(e.g. not requiring a network) allows the user to decide what solution they want to use but ROS2 will still just work, so that's a nice separation of responsibilities. It looks like RTI does implement shared memory as a transport but that Fast-RTPS does not yet, although this page hints that it might become available in the future: 
https://github.com/eProsima/Shapes-Demo-Docs/blob/master/docs/troubleshooting.rst

closed 2017-05-26 by edwinferretjp