ros2/demos #144: talker_qos and talker don't use the same topic name

mikaelarguedas, 2017-06-27:
We can either merge them into a single executable or make them use the same topic (same for listener)

wjwwood, 2017-06-27:
Should they have the same topic?

File links:

- https://github.com/ros2/demos/blob/master/demo_nodes_py/topics/talker.py
- https://github.com/ros2/demos/blob/master/demo_nodes_py/topics/talker_qos.py

Kukanani, 2017-06-27:
I was talking about this with @dhood, and we felt that it was perplexing to be able to run `talker`/`listener` and `talker_qos`/`listener_qos`, but not `talker_qos`/`listener` etc.

Kukanani, 2017-06-27:
^ sorry, `talker_qos`/`listener` is not possible, but `talker`/`listener_qos` should be.

mikaelarguedas, 2017-06-27:
It depends if we want talker to stay as minimal as possible or if we consider that the one in the examples repo serves that purpose. If the latter I think we should just make a single talker/listener that can take a `--reliable` (or a `--best-effort`) argument rather than maintaining 2 versions of it.
Otherwise just matching topic names is fine by me

mikaelarguedas, 2017-06-27:
ok I settled with just making them use the same topic name for now https://github.com/ros2/demos/pull/142/commits/57a13a99c95f7e69cfaa1bfc05bb3ca655a817db

commit 57a13a99 (2017-06-27): make talker_qos and talker use the same topic, fixes #144
PR #142 (2017-06-28): Various fixes
- Install lifecycle demo executables to libexec (fixes #137 )
- install dummy robot/* executables to libexec
- make launchfiles runnable with `launch` (fixes #140 )
- install topic monitor executables to libexec (fixes #138 )
- switch talker/listener_qos to use the chatter topic (fixes #144 )

Not in this PR: 
 - move the pendulum demo executables (because they need to be on the path given the current launch logic)

Depends on https://github.com/ros2/robot_state_publisher/pull/8
Depends on https://github.com/ros2/ros2cli/pull/23
commit a4d48aa5 (2017-06-28): Various fixes (#142)

* install to libexec and update launchfile accordingly

* make lilfecycle_demo_launch.py an actual launchfile

* duh

* install to libexec

* make dummy_robot_bringup.py an actual launchfile

* install topic_monitor executables to libexec

* make talker_qos and talker use the same topic, fixes #144

* use get_executable_path from ros2run

* [lifecycle] install launch folder in share

* [topic_monitor] update topic_monitor README.md to use ros2 run

* [topic_monitor] additional -- for passing args to ros2 run

* [topic_monitor] more readme updates

* [topic_monitor] update [... cut]
closed 2017-06-28 by mikaelarguedas