ros2/demos #152: topic_monitor test seems to fail with debian packages

clalancette, 2017-06-30:
I'm testing out the debian packages on ARM64, and I started following these instructions: https://github.com/ros2/demos/blob/master/topic_monitor/README.md .  However, the very first instruction to run `ros2 run topic_monitor topic_monitor -- --display` ends up in a traceback:

```
Traceback (most recent call last):
  File "/opt/ros/r2b2/lib/topic_monitor/topic_monitor", line 9, in <module>
    load_entry_point('topic-monitor==0.0.0', 'console_scripts', 'topic_monitor')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: No module named 'topic_monitor'
```

dirk-thomas, 2017-06-30:
The Debian package doesn't seem to contain any of the Python modules (at least on amd64):

```
$ dpkg -L ros-r2b2-topic-monitor 
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-r2b2-topic-monitor
/usr/share/doc/ros-r2b2-topic-monitor/changelog.Debian.gz
/opt
/opt/ros
/opt/ros/r2b2
/opt/ros/r2b2/share
/opt/ros/r2b2/share/topic_monitor
/opt/ros/r2b2/share/topic_monitor/package.xml
/opt/ros/r2b2/share/ament_index
/opt/ros/r2b2/share/ament_index/resource_index
/opt/ros/r2b2/share/ament_index/resource_index/packages
/opt/ros/r2b2/share/ament_index/resource_index/packages/topic_monitor
/opt/ros/r2b2/lib
/opt/ros/r2b2/lib/topic_monitor
/opt/ros/r2b2/lib/topic_monitor/topic_monitor
/opt/ros/r2b2/lib/topic_monitor/launch_fragmentation_demo
/opt/ros/r2b2/lib/topic_monitor/launch_reliability_demo
/opt/ros/r2b2/lib/topic_monitor/data_publisher
/opt/ros/r2b2/lib/topic_monitor/launch_depth_demo
/opt/ros/r2b2/lib/python3.5
/opt/ros/r2b2/lib/python3.5/site-packages
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info/dependency_links.txt
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info/top_level.txt
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info/entry_points.txt
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info/PKG-INFO
/opt/ros/r2b2/lib/python3.5/site-packages/topic_monitor-0.0.0.egg-info/requires.txt
```

clalancette, 2017-06-30:
Yeah, the file list looks the same on ARM64.

nuclearsandwich, 2017-06-30:
Examining the [build output](http://build.ros2.org/job/Rbin_uX64__topic_monitor__ubuntu_xenial_amd64__binary/7/consoleFull#console-section-15) now. Looks like it isn't building properly. Now to determine why.

nuclearsandwich, 2017-06-30:
dh_clean/setup.py clean can't find ament_python even during binarydeb time.

```
01:21:00 dh clean -v --buildsystem=pybuild --with python3
01:21:00    dh_testdir -O-v -O--buildsystem=pybuild
01:21:00    debian/rules override_dh_auto_clean
01:21:00 make[1]: Entering directory '/tmp/binarydeb/ros-r2b2-topic-monitor-0.0.0'
01:21:00 # We don't have ament_python in the iamge that creates sourcedebs
01:21:00 # So for now we need to give a pass to dh_auto_clean invoking setup.py clean.
01:21:00 dh_auto_clean || true
01:21:00 	pybuild --clean -i python{version} -p 3.5 --dir .
01:21:00 I: pybuild base:184: python3.5 setup.py clean 
01:21:00 Traceback (most recent call last):
01:21:00   File "setup.py", line 1, in <module>
01:21:00     from ament_python.data_files import get_data_files
01:21:00 ImportError: No module named 'ament_python'
01:21:00 E: pybuild pybuild:274: clean: plugin distutils failed with: exit code=1: python3.5 setup.py clean 
01:21:00 dh_auto_clean: pybuild --clean -i python{version} -p 3.5 --dir . returned exit code 13
```

There's not much output about building the package, but `setup.py build` is invoked

```
01:21:01  debian/rules build
01:21:01 dh build -v --buildsystem=pybuild --with python3
01:21:01    dh_testdir -O-v -O--buildsystem=pybuild
01:21:01    dh_update_autotools_config -O-v -O--buildsystem=pybuild
01:21:01    debian/rules override_dh_auto_configure
01:21:01 make[1]: Entering directory '/tmp/binarydeb/ros-r2b2-topic-monitor-0.0.0'
01:21:01 # In case we're installing to a non-standard location, look for a setup.sh
01:21:01 # in the install tree that was dropped by catkin, and source it.  It will
01:21:01 # set things like CMAKE_PREFIX_PATH, PKG_CONFIG_PATH, and PYTHONPATH.
01:21:01 if [ -f "/opt/ros/r2b2/setup.sh" ]; then . "/opt/ros/r2b2/setup.sh"; fi && \
01:21:01 dh_auto_configure 
01:21:01 	pybuild --configure -i python{version} -p 3.5 --dir .
01:21:02 I: pybuild base:184: python3.5 setup.py config 
01:21:02 running config
01:21:02 make[1]: Leaving directory '/tmp/binarydeb/ros-r2b2-topic-monitor-0.0.0'
01:21:02    debian/rules override_dh_auto_build
01:21:02 make[1]: Entering directory '/tmp/binarydeb/ros-r2b2-topic-monitor-0.0.0'
01:21:02 # In case we're installing to a non-standard location, look for a setup.sh
01:21:02 # in the install tree that was dropped by catkin, and source it.  It will
01:21:02 # set things like CMAKE_PREFIX_PATH, PKG_CONFIG_PATH, and PYTHONPATH.
01:21:02 if [ -f "/opt/ros/r2b2/setup.sh" ]; then . "/opt/ros/r2b2/setup.sh"; fi && \
01:21:02 dh_auto_build
01:21:03 	pybuild --build -i python{version} -p 3.5 --dir .
01:21:03 I: pybuild base:184: /usr/bin/python3 setup.py build 
01:21:03 running build
01:21:03 make[1]: Leaving directory '/tmp/binarydeb/ros-r2b2-topic-monitor-0.0.0'
```

nuclearsandwich, 2017-06-30:
Currently making changes to bloom to deal with https://github.com/ros2/rosdistro/issues/9. I've figured out why ament_python wasn't being found but I'm not sure how or if that would affect the later build.

dirk-thomas, 2017-06-30:
The Debian package of `topic_monitor` does contain its scripts in the libexec folder. So `ament_python` seems to do its job correctly.

nuclearsandwich, 2017-06-30:
The only python file that isn't a script or launch file (which are [registered as entrypoints]( https://github.com/ros2/demos/blob/0e0199665002c0ca72e27d743cb880b1ba251fee/topic_monitor/setup.py#L30-L38) and installed) is the empty file [here](https://github.com/ros2/demos/blob/0e0199665002c0ca72e27d743cb880b1ba251fee/topic_monitor/topic_monitor/__init__.py) correct?

nuclearsandwich, 2017-06-30:
Purging the upstream tarball does indeed resolve the issue. The package currently building on the farm is busted due to the bloom changes around ros2/rosdistro#9 but freshest of fresh packages are blossoming as I write this 🌸

dirk-thomas, 2017-06-30:
Resolved in the latest Debian package (0.0.0-10).

closed 2017-06-30 by dirk-thomas