ros2/demos #159: composition test_api_pubsub_composition with rmw_fastrtps flaky on xenial

sloretz, 2017-08-09:
In the `composition` package the test `test_api_pubsub_composition` is flaky when using `rmw_fastrtps`. When run repeatedly, the test will fail within  1 to 3 minutes.

When the bug occurs:
 *  messages from the publisher are printed to the console window, but no message from the subscriber is ever printed
 * Messages from the publisher are successfully echo'd using `ros2 topic echo`
 * Messages published using `ros2 topic pub` after the bug occurs are either never received by the subscriber, or always received by the subscriber. It seems like there are two kinds failures: one where the subscriber never talks to any publisher, and one where it won't talk to a publisher in the same process.

Output from wireshark during a failing test (starts at message number 25732)
[ros2_composition_pubsub_fail.pcapng.zip](https://github.com/ros2/demos/files/1212705/ros2_composition_pubsub_fail.pcapng.zip)

The following expect script can reproduce the issue (save as `pubsubfail.exp`). The comment at the top can be copy/pasted into bash. It will run the test over and over until the bug is observed, then send SIGINT to drop into a gdb prompt.

```
# while expect pubsubfail.exp $(ros2 pkg executables --full-path composition | grep api_composition\$) ; do date; done

set CTRLC \003
set timeout 10
set api_composition_path [lindex $argv 0]

spawn ros2 run composition api_composition_cli composition composition::Talker
spawn ros2 run composition api_composition_cli composition composition::Listener
spawn gdb -quiet -ex run $api_composition_path

expect {
  "I heard" exit
  timeout {
    send $CTRLC
    interact
  }
}
```

I've been unable to reduce the code. I have only gotten the failure to occur when the talker and listener nodes are loaded in a service callback.

sloretz, 2017-09-07:
This appears to be getting flakier. On September 6th repeated nightlies it failed on all platforms: aarch64, windows, osx, linux. On September 7th repeated nightlies it failed on aarch64, windows, linux. The OSX repeated nightly is still running.

mikaelarguedas, 2017-12-02:
Confirmed, I'm having a lot of trouble to get this to pass 1 out of 10 times locally and most CI need to retry it more than 5 times. I've seen it on Windows several times recently as well.

sloretz, 2018-02-28:
This test is also very flaky on windows debug. Most of the time when I run the following batch script I see the output from the talker but not the listener.

```
start /b D:/ros2/build/composition/Debug/api_composition.exe
D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Talker
D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Listener
```

Failing run example
```
D:\ros2>call D:\ros2\build\composition\test_api_pubsub_composition__rmw_fastrtps_cpp_Debug.bat

D:\ros2>start /b D:/ros2/build/composition/Debug/api_composition.exe

D:\ros2>D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Talker
[INFO] [api_composition_cli]: Sending request...
[INFO] [api_composition_cli]: Waiting for response...
[INFO] [api_composition]: Load library D:\ros2\install\composition/bin/talker_component.dll
[INFO] [api_composition]: Instantiate class composition::Talker
[INFO] [api_composition_cli]: Result of load_node: success = true

D:\ros2>D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Listener
[INFO] [api_composition_cli]: Sending request...
[INFO] [api_composition_cli]: Waiting for response...
[INFO] [api_composition]: Load library D:\ros2\install\composition/bin/listener_component.dll
[INFO] [api_composition]: Instantiate class composition::Listener
[INFO] [api_composition_cli]: Result of load_node: success = true

D:\ros2>[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [talker]: Publishing: 'Hello World: 4'
[INFO] [talker]: Publishing: 'Hello World: 5'
[INFO] [talker]: Publishing: 'Hello World: 6'
[INFO] [talker]: Publishing: 'Hello World: 7'
[INFO] [talker]: Publishing: 'Hello World: 8'
[INFO] [talker]: Publishing: 'Hello World: 9'
[INFO] [talker]: Publishing: 'Hello World: 10'
[INFO] [talker]: Publishing: 'Hello World: 11'
[INFO] [talker]: Publishing: 'Hello World: 12'
[INFO] [talker]: Publishing: 'Hello World: 13'
```

Successful run example
```
D:\ros2>call D:\ros2\build\composition\test_api_pubsub_composition__rmw_fastrtps_cpp_Debug.bat

D:\ros2>start /b D:/ros2/build/composition/Debug/api_composition.exe

D:\ros2>D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Talker
[INFO] [api_composition_cli]: Sending request...
[INFO] [api_composition_cli]: Waiting for response...
[INFO] [api_composition]: Load library D:\ros2\install\composition/bin/talker_component.dll
[INFO] [api_composition]: Instantiate class composition::Talker
[INFO] [api_composition_cli]: Result of load_node: success = true

D:\ros2>D:/ros2/build/composition/Debug/api_composition_cli.exe composition composition::Listener
[INFO] [api_composition_cli]: Sending request...
[INFO] [api_composition_cli]: Waiting for response...
[INFO] [api_composition]: Load library D:\ros2\install\composition/bin/listener_component.dll
[INFO] [api_composition]: Instantiate class composition::Listener
[INFO] [api_composition_cli]: Result of load_node: success = true

D:\ros2>[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [listener]: I heard: [Hello World: 1]
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [listener]: I heard: [Hello World: 2]
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [listener]: I heard: [Hello World: 3]
[INFO] [talker]: Publishing: 'Hello World: 4'
[INFO] [listener]: I heard: [Hello World: 4]
[INFO] [talker]: Publishing: 'Hello World: 5'
[INFO] [listener]: I heard: [Hello World: 5]
[INFO] [talker]: Publishing: 'Hello World: 6'
[INFO] [listener]: I heard: [Hello World: 6]
[INFO] [talker]: Publishing: 'Hello World: 7'
[INFO] [listener]: I heard: [Hello World: 7]
```

mikaelarguedas, 2018-04-10:
Closing this as the fastrtps issue has been resolved upstream (context in https://github.com/ros2/build_cop/issues/86)

closed 2018-04-10 by mikaelarguedas