ros2/demos #160: Listener does not see msg when Talker publishes once

CodingPeon, 2017-08-10:
I'm seeing an issue where listener has a high chance of missing the first msg published to the topic.
I have FastRTPS.

Here is what I see when I run Talker and Listener.

Listener prints:
I heard: [Hello World: 2]
I heard: [Hello World: 3]
Talker prints:
Publishing: 'Hello World: 1'
Publishing: 'Hello World: 2'
Publishing: 'Hello World: 3'

Occasionally Listener does hear all the msgs Talker sends out. But if Talker is executed again while Listener is kept running, Listener misses the first msg again.

dirk-thomas, 2017-08-10:
Can you please clarify which version you are using, a binary release, or from source, and on which platform.

CodingPeon, 2017-08-10:
I'm compiling source with master branch and running on 16.04 LTS

dirk-thomas, 2017-08-10:
I can reproduce the problem with the `demo_nodes_cpp` nodes. With the `talker` from `demo_nodes_py` it seems to work fine though.

The "problem" is that the matching of publishers and subscribers is not *instant* in a distributed system like this. So when the C++ node publishes the first message the publisher isn't aware of the subscriber yet and therefore can't deliver the message. This is less a bug but a conceptional problem in an asynchronous system. You might see the same effect in ROS 1 (see related [question](https://answers.ros.org/question/11167/how-do-i-publish-exactly-one-message/)).

Currently the API doesn't provide a mechanism to indicate if any subscriber has been "connected". Even then you wouldn't necessarily know how many subscribers are expected before your node should start publishing. This should be considered when designing the data flow if your application.

Karsten1987, 2017-08-11:
The problem also further occurs when running in more deterministic fashion. 

```
void
callback(std::shared_ptr<std_msgs::msg::String> msg, bool & callback_received)
{
  (void) msg;
  callback_received = true;
}

TEST_F(TestTalker, inter_talk) {
  auto pub_node = std::make_shared<rclcpp::node::Node>("talker");
  auto sub_node = std::make_shared<rclcpp::node::Node>("listener");

  auto msg = std::make_shared<std_msgs::msg::String>();
  msg->data = "Hello World";

  auto publisher = pub_node->create_publisher<std_msgs::msg::String>("chatter");

  auto callback_received = false;
  std::function<void(std::shared_ptr<std_msgs::msg::String>)> cb_function =
    std::bind(callback, std::placeholders::_1, std::ref(callback_received));
  auto subscription = sub_node->create_subscription<std_msgs::msg::String>("chatter", cb_function);

  rclcpp::executors::SingleThreadedExecutor executor;
  executor.add_node(pub_node);
  executor.add_node(sub_node);

  fprintf(stderr, "publish #1\n");
  publisher->publish(msg);
  fprintf(stderr, "spin once #1\n");
  executor.spin_once();
  fprintf(stderr, "number of subss %zu\n", pub_node->count_subscribers("chatter"));
  fprintf(stderr, "publish #2\n");
  publisher->publish(msg);
  fprintf(stderr, "spin once #2\n");
  executor.spin_once();
  EXPECT_TRUE(callback_received);
}
```

This does lead to a test timeout, printing the following:
```
10: [ RUN      ] TestTalker.inter_talk
10: publish #1
10: spin once #1
10: number of subss 0
10: publish #2
10: spin once #2
1/1 Test #10: test_talker__rmw_fastrtps_cpp ....***Timeout  60.07 sec
```
Where the time out comes from the second call to `executor.spin_once()`. I can't really explain why the first spin would work and the second one timeout. 
The behavior is consistent between fastrtps and connext.

dirk-thomas, 2017-08-11:
@Karsten1987 Your example sounds very similar. Have you tried to add a `sleep` between the entity creation and the publishing of the first message? Also `spin_once` might return for "arbitrary" reasons e.g. an internal event - not only in the case a message was being received.

Karsten1987, 2017-08-31:
closing this issue with respect to the above comments.

closed 2017-08-31 by Karsten1987