ros2/demos #170: Parameter node cross-talk

dhood, 2017-09-11:
If I have a `parameter_node` running, then it spits out errors when I run the `parameter_events` demo in another terminal.

OS X with only Connext 5.2.3, using master.

Terminal A:
```
$ ros2 run demo_nodes_cpp parameter_node
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 09-Oct-2017 See www.rti.com for more information.
```

Terminal B (expected output):
```
$ ros2 run demo_nodes_cpp parameter_events
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 09-Oct-2017 See www.rti.com for more information.
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  bar
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  baz
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  foobar
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:
```

Terminal A:
```
<already started>
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameter_types': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/set_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/describe_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/list_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  bar
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  baz
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
  foobar
 changed parameters:
 deleted parameters:
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:
```

Is the [`on_parameter_event` callback](https://github.com/ros2/demos/blob/79c2f5ac6d904a8a0bb98f5fd23a210d6886682c/demo_nodes_cpp/src/parameters/parameter_node.cpp#L59) supposed to get notified of changes to other nodes?

wjwwood, 2017-09-11:
> Is the on_parameter_event callback supposed to get notified of changes to other nodes?

Yes, at least that's the way it is coded:

- pub: https://github.com/ros2/rclcpp/blob/2c5ab49e7c329fdbfbf761b352572a74f5cee007/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L43
- sub: https://github.com/ros2/rclcpp/blob/1c42a75f439a1ef7c60bda805918b73679daf683/rclcpp/include/rclcpp/parameter_client.hpp#L122

I believe this has been discussed before:

https://github.com/ros2/rclcpp/issues/243

dhood, 2017-09-11:
OK thanks, so if this is the expected communication, then the question is why it's not working cleanly. I'll update the title.

E.g.
```
$ ros2 run demo_nodes_cpp ros2param list parameter_node
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 09-Oct-2017 See www.rti.com for more information.
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameter_types': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/set_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/list_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/describe_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
Node parameter_node has 0 parameters:

$ ros2 run demo_nodes_cpp ros2param set parameter_node/test_param 0
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 09-Oct-2017 See www.rti.com for more information.
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameter_types': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/set_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/list_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/describe_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279

$ ros2 run demo_nodes_cpp ros2param list parameter_node
RTI Data Distribution Service EVAL License issued to OSRF dhood@osrfoundation.org For non-production use only.
Expires on 09-Oct-2017 See www.rti.com for more information.
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/get_parameter_types': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
[rclcpp::error] take response failed for client of service 'parameter_node/set_parameters': error not set, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279, at /Users/dhood/ros2_ws_beta3/src/ros2/rcl/rcl/src/rcl/client.c:279
Node parameter_node has 1 parameters:
test_param
```

The communication appears to be working fine, and I think there's been another instance in the past where a service error was printed when it actually worked, so I will not consider this a high priority yet and will come back to it at the end of testing.

wjwwood, 2017-09-11:
Btw, I don't see this with my setup and/or because of Fast-RTPS (I tried the process you described in the OP).

dhood, 2017-09-12:
not sure why this was happening so repeatedly on my machine (perhaps because I only had connext installed so no dlopen delays) but I've opened https://github.com/ros2/rclcpp/pull/373 and https://github.com/ros2/rclcpp/pull/374 to address it

closed 2017-09-12 by dhood