ros2/demos #172: Compisition wiki page update

mikaelarguedas, 2017-09-12:
@dirk-thomas I'm planning on adding a section for `dlopen_composition` at the end of the [composition tutorial page](https://github.com/ros2/ros2/wiki/Composition). Does this text sound reasonable to you ?
```
### Run-time composition using dlopen

This demo presents an alternative to 1. by creating a generic container process and pass it explicitely 
the libraries to load without using ROS interfaces. The process will open each library and create one 
instance of each "rclcpp::Node" class in the library [source code](https://github.com/ros2/demos/blob/master/composition/src/dlopen_composition.cpp)).

In the shell call:

        ros2 run composition dlopen_composition -- `ros2 pkg prefix composition`/lib/libtalker_component.so `ros2 pkg prefix composition`/lib/liblistener_component.so

Now the shell should show repeated output for each sent and received message.
```

dirk-thomas, 2017-09-12:
LGTM (with wrapping for new sentences).

mikaelarguedas, 2017-09-12:
Thanks for the feedback. added in https://github.com/ros2/ros2/wiki/Composition/_compare/6f50ce099a4417fa41045d1323d2d8dc4e5545fd...5f12eb6cfcd3e32b74b4eccd996170ed79514cf8

closed 2017-09-12 by mikaelarguedas