ros2/demos #175: topic_monitor launch_reliability_demo hangs on exit on Windows

wjwwood, 2017-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - from source
- Version or commit hash:
  - `vcs export --exact`: https://gist.github.com/wjwwood/9e6106e70e8ce624be165c26fa5cfbb1
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run:

```
ros2 run topic_monitor launch_reliability_demo
```

Or:

```
.\install\Lib\topic_monitor\launch_reliability_demo.exe
```

#### Expected behavior

Ends cleanly when ctrl-c.

#### Actual behavior

Hangs on ctrl-c.

#### Additional information

Works as expected on Linux and macOS, and other launch scripts seem to work on Windows.

sloretz, 2017-09-13:
I see the same behavior installing from the archive on both `launch_reliability_demo` and `launch_fragmentation_demo`.

dirk-thomas, 2017-09-13:
The process hangs for me in `launcher.py` on the `loop.close()` call. Both subprocesses have been SIGTERMed and the loop is not running anymore. But the `close()` still hangs forever.

dhood, 2017-09-13:
I think this is a wider launch issue. You can see the same hanging behaviour if you send a keyboard interrupt to a launch script that is used for testing demos, e.g.:
```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC>python C:\Users\osrf\Downloads\r2b3-testing\overlay\build\demo_nodes_cpp\test_add_two_ints_server_add_two_ints_client__rmw_opensplice_cpp_Release.py
(test_executable_0) pid 10048: ['C:/Users/osrf/Downloads/r2b3-testing/overlay/build/demo_nodes_cpp/Release/add_two_ints_server.exe'] (all > console, InMemoryHandler: test_executable_0)
(test_executable_1) pid 4984: ['C:/Users/osrf/Downloads/r2b3-testing/overlay/build/demo_nodes_cpp/Release/add_two_ints_client.exe'] (all > console, InMemoryHandler: test_executable_1)
<manual keyboard interrupt>
(test_executable_0) signal SIGTERM
(test_executable_1) signal SIGTERM
(test_executable_0) rc 1
(test_executable_1) rc 1
<hang>
```

On one of them, however, you get some more info:
```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC>python C:\Users\osrf\Downloads\r2b3-testing\overlay\build\demo_nodes_cpp\test_set_and_get_parameters__rmw_opensplice_cpp_Release.py
(test_executable_0) pid 2276: ['C:/Users/osrf/Downloads/r2b3-testing/overlay/build/demo_nodes_cpp/Release/set_and_get_parameters.exe'] (all > console, InMemoryHandler: test_executable_0)
[test_executable_0] Parameter name: foo
[test_executable_0] Parameter value (integer): 2
[test_executable_0] Parameter name: baz
[test_executable_0] Parameter value (double): 1.450000
<manual keyboard interrupt>
(test_executable_0) rc 1
() tear down
Exception ignored in: <bound method BaseEventLoop.__del__ of <_WindowsSelectorEventLoop running=False closed=False debug=False>>
Traceback (most recent call last):
  File "C:\Python36\lib\asyncio\base_events.py", line 511, in __del__
    self.close()
  File "C:\Python36\lib\asyncio\selector_events.py", line 110, in close
    self._close_self_pipe()
  File "C:\Python36\lib\asyncio\selector_events.py", line 120, in _close_self_pipe
    self._remove_reader(self._ssock.fileno())
  File "C:\Python36\lib\asyncio\selector_events.py", line 279, in _remove_reader
    key = self._selector.get_key(fd)
  File "C:\Python36\lib\selectors.py", line 189, in get_key
    return mapping[fileobj]
  File "C:\Python36\lib\selectors.py", line 70, in __getitem__
    fd = self._selector._fileobj_lookup(fileobj)
  File "C:\Python36\lib\selectors.py", line 224, in _fileobj_lookup
    return _fileobj_to_fd(fileobj)
  File "C:\Python36\lib\selectors.py", line 41, in _fileobj_to_fd
    raise ValueError("Invalid file descriptor: {}".format(fd))
ValueError: Invalid file descriptor: -1
```

I think I remember seeing something in a comment somewhere in launch specifically about file descriptors.. I'll see if I can find it again.

dirk-thomas, 2017-09-13:
@dhood Should this be closed as a duplicate?

dhood, 2017-09-13:
Replaced by https://github.com/ros2/launch/issues/64 which has a minimal example

closed 2017-09-13 by dhood