ros2/demos #177: pendulum_demo sometimes hanging forever

dirk-thomas, 2017-09-15:
The test `test_pendulum_teleop` sometimes hangs forever and therefore the test times out. After reproducing this locally with `RMW_IMPLEMENTATION=rmw_connext_cpp python3 build_isolated/pendulum_control/test_pendulum_teleop__rmw_connext_cpp.py` I was able to look at the stack:

```
#0  __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007f8b6d906dbd in __GI___pthread_mutex_lock (mutex=0x7f8b6c46fc00 <Poco::SharedLibraryImpl::_mutex>) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007f8b6c153a98 in Poco::MutexImpl::lockImpl (this=0x7f8b6c46fc00 <Poco::SharedLibraryImpl::_mutex>) at include/Poco/Mutex_POSIX.h:81
#3  0x00007f8b6c153c86 in Poco::FastMutex::lock (this=0x7f8b6c46fc00 <Poco::SharedLibraryImpl::_mutex>) at include/Poco/Mutex.h:205
#4  0x00007f8b6c1544c0 in Poco::ScopedLock<Poco::FastMutex>::ScopedLock (this=0x7ffc2fe659b0, mutex=...) at include/Poco/ScopedLock.h:59
#5  0x00007f8b6c1bf5e8 in Poco::SharedLibraryImpl::findSymbolImpl (this=0xf39ba8, name="rmw_trigger_guard_condition") at src/SharedLibrary_UNIX.cpp:100
#6  0x00007f8b6c1bf8a1 in Poco::SharedLibrary::hasSymbol (this=0xf39ba0, name="rmw_trigger_guard_condition") at src/SharedLibrary.cpp:93
#7  0x00007f8b6c680839 in get_symbol (symbol_name=0x7f8b6c681f70 "rmw_trigger_guard_condition") at ../ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:148
#8  0x00007f8b6c681076 in rmw_trigger_guard_condition (guard_condition=0xf80e00) at ../ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:423
#9  0x00007f8b6e360963 in rcl_trigger_guard_condition (guard_condition=0x171e4a8) at ../ws_ros2/src/ros2/rcl/rcl/src/rcl/guard_condition.c:154
#10 0x00007f8b6e0c9c8d in trigger_interrupt_guard_condition (signal_value=2) at ../ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/utilities.cpp:115
#11 0x00007f8b6e0c9dc7 in signal_handler (signal_value=2, siginfo=0x7ffc2fe65cf0, context=0x7ffc2fe65bc0) at ../ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/utilities.cpp:158
#12 <signal handler called>
#13 0x00007f8b6c1bf5f8 in Poco::SharedLibraryImpl::findSymbolImpl (this=0xf39ba8, name="rmw_wait") at src/SharedLibrary_UNIX.cpp:103
#14 0x00007f8b6c1bf8a1 in Poco::SharedLibrary::hasSymbol (this=0xf39ba0, name="rmw_wait") at src/SharedLibrary.cpp:93
#15 0x00007f8b6c680839 in get_symbol (symbol_name=0x7f8b6c681fb3 "rmw_wait") at ../ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:148
#16 0x00007f8b6c68114d in rmw_wait (subscriptions=0x7f8b239fb908, guard_conditions=0x7f8b239fb920, services=0x7f8b239fb950, clients=0x7f8b239fb938, waitset=0x1f122f0, wait_timeout=0x7ffc2fe66450)
    at ../ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:456
#17 0x00007f8b6e36b06a in rcl_wait (wait_set=0x1f12878, timeout=0) at ../ws_ros2/src/ros2/rcl/rcl/src/rcl/wait.c:590
#18 0x00007f8b6e021f98 in rclcpp::executor::Executor::wait_for_work (this=0x1f12860, timeout=...) at ../ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:409
#19 0x00007f8b6e0231db in rclcpp::executor::Executor::get_next_executable (this=0x1f12860, timeout=...) at ../ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:544
#20 0x00007f8b6e020b7e in rclcpp::executor::Executor::spin_some (this=0x1f12860) at ../ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:191
#21 0x000000000042124f in pendulum_control::RttExecutor::loop_callback (arg=0x1f12860) at ../ws_ros2/src/ros2/demos/pendulum_control/include/pendulum_control/rtt_executor.hpp:135
#22 0x00007f8b6db2f821 in Rttest::spin_once (this=0xf35a48, user_function=0x4211f5 <pendulum_control::RttExecutor::loop_callback(void*)>, args=0x1f12860, start_time=0x7ffc2fe66750, update_period=0xf35a50, i=1145)
    at ../ws_ros2/src/ros2/realtime_support/rttest/src/rttest.cpp:595
#23 0x00007f8b6db2f576 in Rttest::spin_period (this=0xf35a48, user_function=0x4211f5 <pendulum_control::RttExecutor::loop_callback(void*)>, args=0x1f12860, update_period=0xf35a50, iterations=0)
    at ../ws_ros2/src/ros2/realtime_support/rttest/src/rttest.cpp:551
#24 0x00007f8b6db2f8b8 in rttest_spin_period (user_function=0x4211f5 <pendulum_control::RttExecutor::loop_callback(void*)>, args=0x1f12860, update_period=0xf35a50, iterations=0)
    at ../ws_ros2/src/ros2/realtime_support/rttest/src/rttest.cpp:608
#25 0x00007f8b6db2f4e9 in Rttest::spin (this=0xf35a48, user_function=0x4211f5 <pendulum_control::RttExecutor::loop_callback(void*)>, args=0x1f12860)
    at ../ws_ros2/src/ros2/realtime_support/rttest/src/rttest.cpp:539
#26 0x00007f8b6db2f3f3 in rttest_spin (user_function=0x4211f5 <pendulum_control::RttExecutor::loop_callback(void*)>, args=0x1f12860) at ../ws_ros2/src/ros2/realtime_support/rttest/src/rttest.cpp:512
#27 0x00000000004211bb in pendulum_control::RttExecutor::spin (this=0x1f12860) at ../ws_ros2/src/ros2/demos/pendulum_control/include/pendulum_control/rtt_executor.hpp:95
#28 0x0000000000415d20 in main (argc=3, argv=0x7ffc2fe66c68) at ../ws_ros2/src/ros2/demos/pendulum_control/src/pendulum_demo.cpp:291
```

The relevant pieces of information are in the following frames:
* (15) While checking for the existence of a symbol which locks a mutex
* (12) the incoming `SIGINT` signal is being handled
* (7) which in order to trigger the guard condition needs to check the existence of a different symbol
* :lock:

If the used mutex would be recursive that wouldn't be a problem. But until Poco [1.4.1p1](https://github.com/pocoproject/poco/blob/b9acb9f263d0f3776d87d0de49ded660ab675061/CHANGELOG#L314-L324) Poco uses a non-recursive mutex on Linux. Since Xenial is using version 1.3.6p1 we see the deadlock.

Possible workarounds:
* Use a newer version of Poco (newer Ubuntu distro have a new enough version).
* Build using a single rmw implementation which avoids the usage of Poco in the `rmw_implemenation` package.

mikaelarguedas, 2017-09-15:
what is the behavior of `poco_vendor` if poco is not on the system ? it would download and compile poco from source right? If so that should allow us to use a recent enough version without having to modify code, all we need to do is remove it from the dockerfile and the installation instructions on the wiki

dirk-thomas, 2017-10-06:
And here we go again with a different stacktrace where the thread deadlocks:

```
#0  __lll_lock_wait_private () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:95
#1  0x00007fa8f12f422a in __GI___libc_malloc (bytes=140363581049632) at malloc.c:2911
#2  0x00007fa8f1be6e78 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fa8f1c7aaed in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fa8f1c7ac4c in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fa8f0c5e826 in get_symbol (symbol_name=0x7fa8f0c5ff70 "rmw_trigger_guard_condition") at /home/dthomas/ros2/ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:148
#6  0x00007fa8f0c5f076 in rmw_trigger_guard_condition (guard_condition=0x1bc5670) at /home/dthomas/ros2/ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:424
#7  0x00007fa8f293d963 in rcl_trigger_guard_condition (guard_condition=0x23668e8) at /home/dthomas/ros2/ws_ros2/src/ros2/rcl/rcl/src/rcl/guard_condition.c:154
#8  0x00007fa8f26a704b in trigger_interrupt_guard_condition (signal_value=2) at /home/dthomas/ros2/ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/utilities.cpp:115
#9  0x00007fa8f26a7185 in signal_handler (signal_value=2, siginfo=0x7ffeb6d325b0, context=0x7ffeb6d32480) at /home/dthomas/ros2/ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/utilities.cpp:156
#10 <signal handler called>
#11 0x00007fa8f12f4e6a in __libc_calloc (n=1, elem_size=<optimized out>) at malloc.c:3208
#12 0x00007fa8edf11864 in RTIOsapiHeap_reallocateMemoryInternal (voidPtrPtr=0x7ffeb6d32c68, size=32, alignment=16, reallocFlag=0, forceHeapHeader=0, METHOD_NAME=0x7fa8edf85588 "RTIOsapiHeap_allocateStructure", 
    signature=RTI_OSAPI_STRUCT_ALLOC, typeName=0x7fa8edf855a7 "struct WaitSetNode") at heap.c:748
#13 0x00007fa8eda5a5f1 in PRESCondition_add_waitset (self=0x24d20a8, waitset=0x2b5b2d0, worker=0x1b98b40) at Condition.c:137
#14 0x00007fa8eda5db22 in PRESWaitSet_attach_condition (self=0x2b5b2d0, condition=0x24d20a8, worker=0x1b98b40) at Waitset.c:531
#15 0x00007fa8ee44056a in DDS_WaitSet_attach_condition (self=0x2b5adb0, cond=0x24d2120) at WaitSet.c:531
#16 0x00007fa8ed553d0a in DDSWaitSet_impl::attach_condition (this=0x2b5b1f0, cond=0x24d2178) at WaitSet_impl.cxx:186
#17 0x00007fa8ed5538aa in DDSWaitSet::attach_condition (this=0x2b5b1d0, cond=0x24d2178) at WaitSet.cxx:90
#18 0x00007fa8ef3854c0 in wait<ConnextStaticSubscriberInfo, ConnextStaticServiceInfo, ConnextStaticClientInfo> (implementation_identifier=0x7fa8ef3860a0 "rmw_connext_cpp", subscriptions=0x7fa8b8221908, 
    guard_conditions=0x7fa8b8221920, services=0x7fa8b8221950, clients=0x7fa8b8221938, waitset=0x2b5af70, wait_timeout=0x7ffeb6d33050)
    at /home/dthomas/ros2/ws_ros2/install_isolated/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/wait.hpp:136
#19 0x00007fa8ef384e90 in rmw_wait (subscriptions=0x7fa8b8221908, guard_conditions=0x7fa8b8221920, services=0x7fa8b8221950, clients=0x7fa8b8221938, waitset=0x2b5af70, wait_timeout=0x7ffeb6d33050)
    at /home/dthomas/ros2/ws_ros2/src/ros2/rmw_connext/rmw_connext_cpp/src/rmw_wait.cpp:37
#20 0x00007fa8f0c5f185 in rmw_wait (subscriptions=0x7fa8b8221908, guard_conditions=0x7fa8b8221920, services=0x7fa8b8221950, clients=0x7fa8b8221938, waitset=0x2b5af70, wait_timeout=0x7ffeb6d33050)
    at /home/dthomas/ros2/ws_ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:457
#21 0x00007fa8f294806a in rcl_wait (wait_set=0x2b5aab8, timeout=0) at /home/dthomas/ros2/ws_ros2/src/ros2/rcl/rcl/src/rcl/wait.c:590
...
```

The problem is that `get_symbol(const char * symbol_name)` in `rmw_implementation` calls `Poco::SharedLibrary::hasSymbol` which expects a `const std::string&`. Since the conversion needs to allocate it deadlocks since `calloc` isn't reentrant.

closed 2017-10-06 by dirk-thomas