ros2/demos #212: Multiple clients or servers in composition seems to be flaky

clalancette, 2017-12-07:
## Bug report

I'm currently using Ardent pre-release on ARM64 Linux, with the single-RMW Fast-RTPS implementation.  I'm following the testing guide [here](https://github.com/ros2/ros2/wiki/Composition#run-time-composition-using-ros-services-1-with-a-server-and-client).  If I load up a single Server and Client into the same composition, things work fine.  If I load up a Server and multiple clients into the same composition, then arbitrary-seeming things happen.  In one case, the server started spewing something about an "Invalid sequence number".  In another case, I couldn't successfully shutdown the composition node.  And in a third case, everything worked just fine.

**Required Info:**

- Operating System:
  - Ubuntu 16.04 on ARM64
- Installation type:
  - Fat binary, Fast-RTPS-only build
- Version or commit hash:
  - demos git hash 40e09c17db74c841a43c9ca05a17bb8011a6eef7
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

In terminal one:

```
ros2 run composition api_composition
```

In terminal two:

```
ros2 run composition api_composition_cli -- composition composition::Server
ros2 run composition api_composition_cli -- composition composition::Client
ros2 run composition api_composition_cli -- composition composition::Client
```

#### Expected behavior

With a single Server and two Clients, it seems like they should just happily alternate getting results from the server.

#### Actual behavior

Somewhat different results every time, ranging from server not being ready to being unable to shut down the composition node, to working just fine.

dhood, 2017-12-07:
Probably related to https://github.com/ros2/rclcpp/issues/392 since this demo uses a single-threaded executor

clalancette, 2017-12-15:
@mikaelarguedas Dug into this a bit after I first opened it, and he thinks that the problem has to do with the clients having non-unique names.  We should probably get to the bottom of it for sure to determine whether this is a bug, or just the way it is with composition.

mikaelarguedas, 2017-12-15:
Just to clarify: I didnt look into it at all, I just said that this breaks the unique node name assumption and thus is undefined behavior. But as @dhood pointed out, there are known issues that can be at play here as well. We should try to have a similar setup but with several clients with different names (e.g. create a composition::Client2 component connecting to the same server) and from there investigate the behavior as I'm not convinced the names are the root cause of the observed flakiness.

mikaelarguedas, 2018-02-22:
This needs a reproducible example with a supported use case

dirk-thomas, 2018-02-22:
I removed the "untargeted" milestone since the "more-information-needed" label already expresses that we are waiting for additional information. But since we want this to be addressed "untargeted" seems inappropriate to me (meaning we don't want to look at it anytime).

matt-attack, 2018-03-02:
I have been seeing somewhat similar issues to this involving composition. Possibly related to https://github.com/ros2/rclcpp/issues/447 and some kind of thread safety/race condition issue when there are multiple nodes in a process.

clalancette, 2019-05-30:
The api_composition demo no longer exists, so I'm going to close this issue out.  If anything similar comes up with the "rclcpp_components" replacement (which I haven't seen), we can deal with it in a separate ticket.

closed 2019-05-30 by clalancette

mjcarroll, 2019-05-30:
Correct, the `api_composition` demo has been superseded in `Dashing`-forward by first-class support of components via `rclcpp_components` and the corresponding command line tooling `ros2 component`.