ros2/demos #213: Trying to get the state of lifecycle listener doesn't work

clalancette, 2017-12-07:
## Bug report

I'm running the tutorial at https://github.com/ros2/ros2/wiki/managed-nodes#run-the-demo on Linux ARM64 with the Fast-RTPS only fat archive.  If I run the `lifecycle_talker`, then try `get_state` with `lifecycle_service_client_py.py`, I get the correct state.  If I then run `lifecycle_listener`, and try to get the state using `lifecycle_service_client_py.py`, it hangs forever.  When I eventually hit Ctrl-C, it throws a traceback:

```
^CTraceback (most recent call last):
  File "/home/ubuntu/ros2-linux/lib/lifecycle/lifecycle_service_client_py.py", line 125, in <module>
    main(args.service, args.node, args.change_state_args)
  File "/home/ubuntu/ros2-linux/lib/lifecycle/lifecycle_service_client_py.py", line 104, in main
    get_state(lifecycle_node)
  File "/home/ubuntu/ros2-linux/lib/lifecycle/lifecycle_service_client_py.py", line 58, in get_state
    % (lifecycle_node, cli.response.current_state.label, cli.response.current_state.id))
AttributeError: 'NoneType' object has no attribute 'current_state'
Exception ignored in: <bound method Node.__del__ of <rclpy.node.Node object at 0x7fa6622860>>
Traceback (most recent call last):
  File "/home/ubuntu/ros2-linux/lib/python3.5/site-packages/rclpy/node.py", line 300, in __del__
    self.destroy_node()
  File "/home/ubuntu/ros2-linux/lib/python3.5/site-packages/rclpy/node.py", line 275, in destroy_node
    _rclpy.rclpy_destroy_node_entity(cli.client_handle, self.handle)
RuntimeError: Failed to fini 'rcl_client_t': rcl node is invalid, rcl instance id does not match, at /home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl/src/rcl/node.c:382
```

**Required Info:**

- Operating System:
  - Ubuntu 16.04 ARM64
- Installation type:
  - Fast-RTPS-only fat archive
- Version or commit hash:
  - c44e525cb9c3c13b3f5dd04284e28f0d76b36b55 of demos repo
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp/rclpy

#### Steps to reproduce issue

Terminal 1:
```
ros2 run lifecycle lifecycle_talker
```

Terminal 2:
```
ros2 run lifecycle lifecycle_listener
```

Terminal 3:
```
ros2 run lifecycle lifecycle_service_client_py.py -- get_state lc_listener
```

#### Expected behavior

Should be able to get the current lifecycle state of lc_listener.

#### Actual behavior

The call to `lifecycle_service_client_py.py` hangs forever.  If I Ctrl-C, I get the stacktrace above.

mikaelarguedas, 2017-12-07:
Looking at that demo it looks like expected behavior.
The lifecycle_listener is not a lifecycle node, just a node: https://github.com/ros2/demos/blob/c44e525cb9c3c13b3f5dd04284e28f0d76b36b55/lifecycle/src/lifecycle_listener.cpp#L34
So it doesnt have a get state service. So the call hangs forever.

What we can do about it:
- Modifiy ros2 service call to timeout rather than waiting for service forever
- Make it clear that the demo it just to show the use of a lifecycle talker and that the listener is NOT a lifecycle node

Karsten1987, 2017-12-07:
Where it is indeed the expected behavior, I agree, it's certainly not user friendly. #214 introduces a call to wait_for_service which would eventually time out when the services are not available and thus cleanup nicely.

closed 2017-12-08 by Karsten1987