ros2/demos #226: use of Context in intra process demos

sagniknitr, 2018-01-16:
Suppose I modify the two node pipeline such that there are now four nodes but one pair have a common context and the others don't. Then by the defination of using context the two pairs will not share their messages.
`
auto producer = std::make_shared<Producer>("producer","number");`
`auto consumer = std::make_shared<Consumer>("consumer", "number");
 `
`auto context_test = std::make_shared<rclcpp::Context>();`
` 
auto producer2 = std::make_shared<Producer2>("producer2","number",context_test);`
 `auto consumer2 = std::make_shared<Consumer2>("consumer2", "number",context_test); `

However that is not the case when I am using the above definition in a modified two_nodes_pipeline source code, as both the pair of nodes are receiving each other messages.Is there something I am doing wrong ?

dirk-thomas, 2018-01-16:
The context is being used for the intra-process communication. The publishers and subscribers not in the same context will still exchange messages through the local loopback interface.

wjwwood, 2018-01-16:
👍 to what @dirk-thomas said, this is the right answer. The context is only used to share resources between nodes, for example the intraprocess manager which allows message data to be passed to other entities within the same context without going through DDS. However, if they don't share the context, then they will still exchange data, but it will instead go through serialization/deserialization and through DDS (not necessarily going through local loopback, but possibly doing that too).

dirk-thomas, 2018-02-01:
I assume our comments answered your question hence I will close the ticket for now. Feel free to continue commenting on it.

closed 2018-02-01 by dirk-thomas