ros2/demos #248: image_pipeline_all_in_one doesn't work with FastRTPS

dirk-thomas, 2018-06-20:
The `CameraNode` is publishing images but the `WatermarkNode` never receives any messages. As a consequence the image view GUI is not even shown.

The problem is the same even when the `use_intra_process_comms` flag is set to `false` for the three nodes.

With Connext and OpenSplice the demo works flawless.

mikaelarguedas, 2018-06-20:
Can you `ros2 topic echo` the images published by the CameraNode?
I havent been able to echo them on Xenial

mikaelarguedas, 2018-06-21:
After some investigation, the problem is not related to this demo but to any best_effort publisher and was introduced by https://github.com/eProsima/Fast-RTPS/commit/f8e3abf1c2790457c3e995a18b28a029c3d19793#diff-36c4b31ccdbe8d22a765182dad3a19a8.
This can easily be reproduced by (not working):
```
ros2 run demo_nodes_py talker_qos
ros2 run demo_nodes_py listener_qos
```

And making the publisher reliable works:
```
ros2 run demo_nodes_py talker_qos --reliable
ros2 run demo_nodes_py listener_qos
```

Ticketed upstream at https://github.com/eProsima/Fast-RTPS/issues/230

dirk-thomas, 2018-06-22:
Addressed by the upstream fix.

closed 2018-06-22 by dirk-thomas