ros2/demos #249: Launch file for lifecycle_demo_launch.py can't be used

clalancette, 2018-06-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic arm64
- Installation type:
  - Fat archive from 2018-06-21, downloaded from here: https://ci.ros2.org/view/packaging/job/packaging_linux-aarch64/475/artifact/ws/ros2-package-linux-aarch64.tar.bz2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ ros2 launch ./share/lifecycle/launch/lifecycle_demo_launch.py
```

#### Expected behavior

The lifecycle demo launches.

#### Actual behavior

```
Traceback (most recent call last):
  File "/root/ros2-linux/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.4.0', 'console_scripts', 'ros2')()
  File "/root/ros2-linux/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/root/ros2-linux/lib/python3.6/site-packages/ros2launch/command/launch.py", line 78, in main
    debug=args.debug
  File "/root/ros2-linux/lib/python3.6/site-packages/ros2launch/api/api.py", line 122, in launch_a_python_launch_file
    launch_description = get_launch_description_from_python_launch_file(python_launch_file_path)
  File "/root/ros2-linux/lib/python3.6/site-packages/ros2launch/api/api.py", line 109, in get_launch_description_from_python_launch_file
    python_launch_file_path, 'generate_launch_description()'
ros2launch.api.api.InvalidPythonLaunchFileError: launch file at './share/lifecycle/launch/lifecycle_demo_launch.py' does not contain the required function 'generate_launch_description()'
```

dhood, 2018-06-21:
That file uses "legacy" launch and hasn't been converted for use with `ros2 launch` yet. Did you find reference to that invocation in a tutorial, or were you just inspired because of the name (which would be reasonable also)?

wjwwood, 2018-06-21:
(also the recommended syntax for `ros2 launch` is `ros2 launch <package name> <launch_file.launch.py>`)

clalancette, 2018-06-21:
> That file uses "legacy" launch and hasn't been converted for use with ros2 launch yet. Did you find reference to that invocation in a tutorial, or were you just inspired because of the name (which would be reasonable also)?

Just inspired by the name; I went looking for things that I could launch, and happened on this one :).  This isn't a priority to fix for bouncy, I just thought I would open the bug while I was here testing.

dhood, 2018-06-21:
Yes, I doubt you would be the only person to be misled

Karsten1987, 2018-06-21:
I can second that. Invoking the launch file with the classical `launch` command works well for me.

wjwwood, 2018-06-21:
If you leave this open, then I'd say you should change the title, because the launch file can be used (with the old launch command), instead there should be an issue about converting this launch file to the new launch format or something like that.

clalancette, 2018-06-21:
I'll update the title.

wjwwood, 2018-06-26:
I added a better error message for this case in https://github.com/ros2/launch/pull/93

mikaelarguedas, 2018-07-20:
fixed in #262

@ironmig Tip: you can use the github keywords "fixes" or "closes" if you want issue to be closed automatically when merging.

closed 2018-07-20 by mikaelarguedas