ros2/demos #255: showimage could not deserialize message when cam2image interrupted

sloretz, 2018-06-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1042/
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
In one terminal
```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run image_tools showimage
```
in another
```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run image_tools cam2image -b
```

Then in the `cam2image` terminal hit `CTRL` + `c`

#### Expected behavior

showimage would stop receiving images and nothing would be printed to the console.

#### Actual behavior

`showimage` executable outputs this message when `cam2image` is stopped.
```
cdr stream doesn't contain data
deserialize from cdr buffer failed
[ERROR] [rclcpp]: could not deserialize serialized message on topic '/image': can't convert cdr stream to ros message, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_connext\rmw_connext_cpp\src\rmw_take.cpp:185, at C:\J\workspace\packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\subscription.c:258
```

#### Additional information

Not sure if this belongs here or in the `rmw_connext_cpp` repo

Karsten1987, 2018-06-22:
I can reproduce this behavior on OSX as well.

Karsten1987, 2018-06-22:
It's essentially failing here in `rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em`

```
  if (@(spec.base_type.type)_Plugin_deserialize_from_cdr_buffer(
      dds_message,
      cdr_stream->buffer,
      static_cast<unsigned int>(cdr_stream->buffer_length)) != RTI_TRUE)
  {
    fprintf(stderr, "deserialize from cdr buffer failed\n");
    return false;
  }
```

I am not sure if this is a real bug in connext or not (the API is not part of `rmw_connext_cpp`). I believe this might be some race condition where a callback is triggered but no data is delivered with it. Do you except some better exception handling?

dirk-thomas, 2018-06-27:
Addressed by ros2/rmw_connext#297.

closed 2018-06-27 by dirk-thomas