ros2/demos #261: no stdout output from parameter_events using connext

sloretz, 2018-06-25:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_windows/1046/
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run demo_nodes_cpp parameter_events
```

#### Expected behavior

Should print text about parameters
```
[INFO] [parameter_events]:
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:

[INFO] [parameter_events]:
Parameter event:
 new parameters:
  bar
 changed parameters:
 deleted parameters:

[INFO] [parameter_events]:
Parameter event:
 new parameters:
  baz
 changed parameters:
 deleted parameters:

[INFO] [parameter_events]:
Parameter event:
 new parameters:
  foobar
 changed parameters:
 deleted parameters:

[INFO] [parameter_events]:
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:

[INFO] [parameter_events]:
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:

```

#### Actual behavior

Prints license message and exits
```
>ros2 run demo_nodes_cpp parameter_events
RTI Data Distribution Service Evaluation License issued to OSRF (OSRF01) dthomas@osrfoundation.org For non-production use only.
Expires on 5-Nov-2018 See www.rti.com for more information.

```

#### Additional information

The expected behavior occurs using fastrtps and opensplice. `parameter_events_async` with connext also works fine.

ironmig, 2018-06-26:
Seems to be exclusive to windows, could **not** replicate on Bionic or OSX.

mikaelarguedas, 2018-06-27:
I tried from source and from binaries on windshield and could not reproduce the issue.
I'll try in a recently setup VM to see if there's any difference

mikaelarguedas, 2018-06-28:
Could not reproduce in a fresh vm either

sloretz, 2018-07-11:
Retested using https://ci.ros2.org/view/packaging/job/packaging_windows/1060/ and I still see the issue in my VM, but I'm unable to reproduce on `windows_portable`.

sloretz, 2018-09-06:
Closing this assuming it is not an issue on a more recent version of windows.

closed 2018-09-06 by sloretz