ros2/demos #291: Handle KeyboardIntterrupt in Python nodes

hidmic, 2018-12-11:
## Feature request
Python demos output would be much cleaner if a stacktrace is not printed when a `KeyboardInterrupt` is thrown.

jubeira, 2018-12-14:
Just to add one more case to take into account in this category, I saw a different error message with RMW = `connext` when sending a `KeyboardInterrupt` to `add_two_ints_client` after a service request had been processed, but before the script exists by itself:

```
Failed to trigger guard_condition: guard condition handle is null, at /tmp/binarydeb/ros-crystal-rmw-connext-shared-cpp-0.6.1/src/trigger_guard_condition.cpp:29, at /tmp/binarydeb/ros-crystal-rcl-0.6.3/src/rcl/guard_condition.c:160

The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/opt/ros/crystal/lib/demo_nodes_py/add_two_ints_client", line 11, in <module>
    load_entry_point('demo-nodes-py==0.6.1', 'console_scripts', 'add_two_ints_client')()
  File "/opt/ros/crystal/lib/python3.6/site-packages/demo_nodes_py/services/add_two_ints_client.py", line 38, in main
    node.destroy_node()
  File "/opt/ros/crystal/lib/python3.6/site-packages/rclpy/node.py", line 422, in destroy_node
    _rclpy.rclpy_destroy_entity(self.handle)
SystemError
>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:
  'node handle is null, at /tmp/binarydeb/ros-crystal-rmw-connext-shared-cpp-0.6.1/src/node.cpp:414, at /tmp/binarydeb/ros-crystal-rcl-0.6.3/src/rcl/node.c:490'
```

Again, this seems to be just a particular case when handling `KeyboardInterrupt`.

wjwwood, 2018-12-14:
@jubeira that looks like a different issue (from catching `KeyboardInterrupt`), it looks like a "use after free" problem (or a "use before initialized" issue).

I think it would make sense to create a separate issue for that, if you have time.

clalancette, 2024-07-26:
I ended up doing this as part of https://github.com/ros2/rclpy/issues/1280

closed 2024-07-26 by clalancette