ros2/demos #292: [allocator_tutorial] More global allocations used when intra-process is enabled using Fast_RTPS

sloretz, 2018-12-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic
- Installation type:
  - debs
- Version or commit hash:
  - crystal pre-release
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

sloretz, 2018-12-11:
Labeled as question because I'm not sure what the correct expected behavior is, but the actual behavior is unexpected to me.

wjwwood, 2018-12-20:
I don't know for sure what should be expected here, but I can say that the intra process implementation does do memory allocations depending on what signature of publish is called by the user.