ros2/demos #293: talker/listener nodes with different DDS implementations are emitting warnings/errors

thomas-moulard, 2018-12-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Fat archive
- Version or commit hash:
  - ROS 2 Crystal Clemmys - Pre-release
    https://github.com/ros2/ros2/releases/download/prerelease-crystal-2018-12-08/ros2-package-linux-bionic-x86_64.tar.bz2
- DDS implementation:
  - Multiple combinations
- Client library (if applicable):
  - talker, listener (both cpp and py versions)

#### `fastrtps` -> `opensplice`

When terminating the listener node by pressing CTRL+C, some errors appear in the **publisher** node.

No error appear if the publisher node terminates first.

```
$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run demo_nodes_cpp talker 
[...]
[INFO] [talker]: Publishing: 'Hello World: 26'
[INFO] [talker]: Publishing: 'Hello World: 27'
[INFO] [talker]: Publishing: 'Hello World: 28'
[INFO] [talker]: Publishing: 'Hello World: 29'
[INFO] [talker]: Publishing: 'Hello World: 30'
[INFO] [talker]: Publishing: 'Hello World: 31'
[INFO] [talker]: Publishing: 'Hello World: 32'

**** those messages appear when terminating the listener node
2018-12-11 22:37:51.465 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.466 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.467 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.468 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.469 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.469 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.469 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.469 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.470 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.470 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.470 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.470 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.470 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.471 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.471 [RTPS_MSG_IN Error] (ID:140439302366976) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
2018-12-11 22:37:51.472 [RTPS_MSG_IN Error] (ID:140439310759680) Bad encapsulation for KeyHash and status parameter list -> Function proc_Submsg_Data
[INFO] [talker]: Publishing: 'Hello World: 33'
```

```
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run demo_nodes_cpp listener
[INFO] [listener]: I heard: [Hello World: 25]
[INFO] [listener]: I heard: [Hello World: 26]
[INFO] [listener]: I heard: [Hello World: 27]
[INFO] [listener]: I heard: [Hello World: 28]
[... 68 lines cut ...]
[WARN] [rmw_connext_shared_cpp]: unique topic attempted to be added twice, ignoring
[INFO] [listener]: I heard: [Hello World: 1]
[INFO] [listener]: I heard: [Hello World: 2]
[INFO] [listener]: I heard: [Hello World: 3]
[INFO] [listener]: I heard: [Hello World: 4]
[INFO] [listener]: I heard: [Hello World: 5]
[INFO] [listener]: I heard: [Hello World: 6]
```

#### `connext` -> `fastrtps`

The `connext` talker node is printing out the same message "Could not remove remote endpoint" than the `fastrtps` -> `connext` scenario.
In this case, the `talker` prints out the error message.

The error message is printed out independently of whether the talker or listener is terminated first.

#### `connext` -> `opensplice`

Talker prints out from time to times:

```
[WARN] [rmw_connext_shared_cpp]: unique topic attempted to be added twice, ignoring
```

Behavior seems identical to `opensplice` -> `connext` scenario.

MiguelCompany, 2018-12-14:
For `fastrtps` <-> `connext` cases, we found Connext could be too restrictive about part of the guidPrefix. We solved the error output problems on eProsima/Fast-RTPS#353, that has just been merged on master.

For `fastrtps` <-> `opensplice` cases, it has been reported on eProsima/Fast-RTPS#142. Fast-RTPS does not yet support key-only data submessages. Opensplice sends key-only data messages to inform its endpoints dissapeared.

MiguelCompany, 2019-02-12:
For `fastrtps` <-> `opensplice` cases, this has been solved by eProsima/Fast-RTPS#371, which will be included on release 1.7.1 (eProsima/Fast-RTPS#393).