ros2/demos #294: lifecycle demo broken in Crystal pre-release: service client fails on get_state

thomas-moulard, 2018-12-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Crystal pre-release fat archive
- Version or commit hash:
  - https://github.com/ros2/ros2/releases/tag/prerelease-crystal-2018-12-08
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
~/ros2-linux$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 launch lifecycle lifecycle_demo.launch.py
[INFO] [launch]: process[lifecycle_talker-1]: started with pid [47151]
[INFO] [launch]: process[lifecycle_listener-2]: started with pid [47152]
[INFO] [launch]: process[lifecycle_service_client-3]: started with pid [47153]
[INFO] [lc_listener]: notify callback: Transition from state unconfigured to configuring
[INFO] [lc_listener]: notify callback: Transition from state configuring to inactive
[INFO] [lc_talker]: on_configure() is called.
[INFO] [lc_client]: Transition 1 successfully triggered.
[INFO] [lc_client]: Node lc_talker has current state inactive.
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[WARN] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[ERROR] []: Unable to start transition 1 from current state inactive: Transition is not registered., at /home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:321
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
[INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [lc_client]: Service /lc_talker/change_state is not available.
[INFO] [launch]: process[lifecycle_talker-1]: process has finished cleanly
[INFO] [launch]: process[lifecycle_listener-2]: process has finished cleanly
[INFO] [launch]: process[lifecycle_service_client-3]: process has finished cleanly
[INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [rclcpp]: signal_handler(signal_value=2)
```

```
~/ros2-linux$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run lifecycle lifecycle_service_client get_state lc_talker
[WARN] [lc_client]: Failed to trigger transition 1
***** hangs forever, need to CTRL-C to terminate the process
```

#### Expected behavior

client should not hang, client should display the following:

```
$ ros2 run lifecycle lifecycle_service_client_py.py get_state lc_talker
lc_talker is in state unconfigured(1)
```

Output from https://index.ros.org/doc/ros2/Managed-Nodes/

#### Actual behavior

- hangs
- displays an error message `[WARN] [lc_client]: Failed to trigger transition 1`

Karsten1987, 2018-12-12:
If I look at your first output, the lifecycle node was `ctrl-c` which means that the attached services are no longer available. 
I agree that the script should exit cleanly in this case, but I doubt that the state is correctly returned, rather than something saying "service not available".

Karsten1987, 2018-12-12:
Also the python script `lifecycle_service_client_py.py` should no longer be available in crystal. But you're right, it's still referenced in the documentation.

`ros2 run lifecycle lifecycle_service_client` is a demo like script in which all possible states of a lifecycle machine are getting triggered in order. It doesn't take any additional command line arguments. 
In order to get the state of a running lifecycle node, you can run
```
ros2 lifecycle get <node_name>
```

thomas-moulard, 2018-12-12:
I see, thanks for the explanation.

Still, `ros2 run lifecycle lifecycle_service_client get_state lc_talker` ...is hanging, is this the expected behavior? Is this binary deprecated, if yes, should it work or should it be removed?

Additionally, after having launched `ros2 launch lifecycle lifecycle_demo.launch.py`, I have the following:
```
$ ros2 lifecycle get lc_talker
Node not found
$ ros2 lifecycle list lc_talker
- configure [1]
        Start: unconfigured
        Goal: configuring
- shutdown [5]
        Start: unconfigured
        Goal: shuttingdown
$ ros2 lifecycle get lc_talker
Node not found
```
How come `lifecycle list` finds the node but not `get` ?

launch file log:
```
[INFO] [launch]: process[lifecycle_talker-1]: started with pid [71502]
[INFO] [launch]: process[lifecycle_listener-2]: started with pid [71503]
[INFO] [launch]: process[lifecycle_service_client-3]: started with pid [71504]
[ERROR] [lc_client]: Server time out while getting current state for node lc_talker
```

thomas-moulard, 2018-12-12:
SGTM, renamed the issue and will update the spreadsheet.

sloretz, 2018-12-13:
I think this can be closed now that ros2/ros2_documentation#47 is merged.

closed 2018-12-13 by sloretz