ros2/demos #295: [dummy_robot] rviz intermmittently displays lidar data: Lookup would require extrapolation into the future

sloretz, 2018-12-12:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu bionic
- Installation type:
  - debs
- Version or commit hash:
  - crystal prerelease
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Launch the dummy robot launch file
   ```
   ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
   ```
1. Launch rviz, switch to `world` frame, add a tf and laser scan display

#### Expected behavior

Lidar data should be displayed smoothly.

#### Actual behavior

rviz logs a message from tf about a lookup requiring extrapolation into the future for many of the lidar scans.

```
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.05575 but the latest data is at time 1544576197.04664, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.08754 but the latest data is at time 1544576197.06661, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.12078 but the latest data is at time 1544576197.10727, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.15436 but the latest data is at time 1544576197.14733, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.18861 but the latest data is at time 1544576197.16758, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.22129 but the latest data is at time 1544576197.20804, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.25517 but the latest data is at time 1544576197.23021, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1544576197.28887 but the latest data is at time 1544576197.26705, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
```

The scan displays intermittently
[image]

#### Additional information

I'm not sure what the correct fix is. Maybe rviz is handling the lidar subscription sooner than the tf frames? If so, maybe rviz should try to re-display the lidar data when the tf frame comes in? Or maybe the dummy robot demo should future date the transforms?

wjwwood, 2018-12-12:
Likely a duplicate of https://github.com/ros2/rviz/issues/332, but either way it's almost certainly (but not 100% sure) an issue with rviz or tf, but not the demo.

sloretz, 2018-12-12:
> Likely a duplicate of ros2/rviz#332

Closing this one

closed 2018-12-12 by sloretz