ros2/demos #297: Intra Process Communication demo stalls on Q or Esc key press

gonzodepedro, 2018-12-12:
## Bug report

**Required Info:**

- Operating System:
   - Ubuntu 18.04
- Installation type:
  - binaries (.deb)
- Version or commit hash:
  - Crystal
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Run intra process demo: 
  ros2 run intra_process_demo image_pipeline_all_in_one
Press Esc or Q key

#### Expected behavior
 Application should exit

#### Actual behavior
 Application stalls

#### Additional information

No error on console. Application stops responding to commands. Application exits without errors with SIGINT

wjwwood, 2018-12-12:
@hidmic could this be related to our changes? Did we try a self exiting application?

wjwwood, 2018-12-12:
Oh this is with debs, maybe not then. Also this is rclcpp not rclpy as the issue template says. Unless it is hanging in ros2 run.

hidmic, 2018-12-12:
This might be a duplicate of #290. If nothing is published on the topic, the viewer becomes unresponsive to user input. @gonzodepedro could you validate this?

gonzodepedro, 2018-12-12:
Been testing a little further. It doesn't stalls, but it takes a lot (around 30 seconds) to exit.

wjwwood, 2018-12-12:
Connext is known to take a long time to shutdown, but not that long (30 seconds). Compare with ctrl-c on something like `ros2 run demo_nodes_cpp talker`.

gonzodepedro, 2018-12-12:
> This might be a duplicate of #290. If nothing is published on the topic, the viewer becomes unresponsive to user input. @gonzodepedro could you validate this?

I don't think it's a duplicate of 290

gonzodepedro, 2018-12-12:
> Connext is known to take a long time to shutdown, but not that long (30 seconds). Compare with ctrl-c on something like `ros2 run demo_nodes_cpp talker`.

Takes longer than SIGINT on `ros2 run demo_nodes_cpp talker`.

wjwwood, 2018-12-12:
Running with the command line argument `__log_level:=debug` might shed some light.

jacobperron, 2018-12-14:
Also happens on OSX (High Sierra) with the latest binaries.