ros2/demos #301: [lifecycle] Lifecycle service client needs improvement

hidmic, 2018-12-13:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu Bionic 18.04 (as in [Nvidia OpenGL Docker image](https://hub.docker.com/r/nvidia/opengl/))
- Installation type:
  - Binary (debs)
- Version or commit hash:
  - Crystal pre-release.
- DDS implementation:
  - ADLink OpenSplice 6.9
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Launch lifecycle demo:
```sh
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 launch lifecycle lifecycle_demo.launch.py 
```
Now, attempt to interrupt it with Ctrl+C.

#### Expected behavior
Nodes exit gracefully and so does the CLI.
#### Actual behavior
Lifecycle service client node does not exit gracefully and SIGINT quickly escalates to SIGTERM:

```sh
[ERROR] [launch]: process[lifecycle_service_client-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [launch]: sending signal 'SIGTERM' to process[lifecycle_service_client-3]
[ERROR] [launch]: process[lifecycle_service_client-3] process has died [pid 16138, exit code -15, cmd '/opt/ros/crystal/lib/lifecycle/lifecycle_service_client'].
```

#### Additional information
The `lifecycle_service_client` node uses regular `std::thread_sleep_for` with no regard of the state of `rclcpp::ok()` [here](https://github.com/ros2/demos/blob/master/lifecycle/src/lifecycle_service_client.cpp#L202).

clalancette, 2019-05-30:
@hidmic I believe this is fixed in Dashing; can you confirm?

hidmic, 2019-06-06:
I can confirm!

closed 2019-06-06 by hidmic