ros2/demos #337: Parameter service demo fails on macOS with OpenSplice

jacobperron, 2019-05-15:
## Bug report

**Required Info:**

- Operating System:
  - macOS
- Installation type:
  - source
- Version or commit hash:
  - d57a917c4ecd26431160277af6bc256167450382
- DDS implementation:
  - OpenSplice
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Build and run tests for `demo_nodes_cpp` on macOS with OpenSplice.

#### Expected behavior

Parameter service related tests pass.

#### Actual behavior

Tests fail. See example CI: https://ci.ros2.org/job/ci_osx/5821/testReport/

The test failures appear to be caused by a segmentaion fault in the parameter service node.
We see this by running the demos manually:
```
$ build/demo_nodes_cpp/list_parameters
[INFO] [list_parameters]: Setting parameters...
Segmentation fault: 11
```
#### Additional information

This test failure was first noticed after changes related to https://github.com/ros2/ros2/issues/677.
Given the nature of the change, it seems more likely that it uncovered an existing issue rather than introducing the bug directly. Bear in mind that this is all speculation.

jacobperron, 2019-05-16:
I've identified that the segfault is occurring because of a dynamic cast that is failing in the generated file `SetParameters_Dcps.cpp:453`:

```cpp
frame #0: 0x0000000101dffbc6 librcl_interfaces__rosidl_typesupport_opensplice_c.dylib`rcl_interfaces::srv::dds_::Sample_SetParameters_Request_DataWriter::_narrow(p=0x0000000103ac0db0) at SetParameters_Dcps.cpp:453
   451 	   if (p && p->_is_a (rcl_interfaces::srv::dds_::Sample_SetParameters_Request_DataWriter::_local_id))
   452 	   {
-> 453 	      result = dynamic_cast < rcl_interfaces::srv::dds_::Sample_SetParameters_Request_DataWriter_ptr> (p);
   454 	      if (result) result->m_count++;
   455 	   }
   456 	   return result;
   457 	}
```

The result of the dynamic cast is `nullptr`, which is not guarded against by [the calling code](https://github.com/ros2/rosidl_typesupport_opensplice/blob/8b0a298f9832199eff0f00575359b587cf614604/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.em#L194).

Even if we guard against the failed cast (which we should anyways), it won't solve this issue.
AFAICT, there is no reason for the dynamic cast to fail. We're casting a `DDS::DataWriter *` to a `rcl_interfaces::srv::dds_::Sample_SetParameters_Request_DataWriter *`, which I see is a derived class by looking at the generated code:

```c++
class Sample_SetParameters_Request_DataWriter
:
    virtual public DDS::DataWriter
{
...
```

dirk-thomas, 2019-05-16:
Is this the same as ros2/rmw_opensplice#264?

jacobperron, 2019-05-16:
> Is this the same as ros2/rmw_opensplice#264?

Maybe, but I didn't notice these specific errors with the parameter demos before https://github.com/ros2/rosidl_typesupport_opensplice/pull/28 was introduced. I could have missed it though.

jacobperron, 2019-05-16:
Indeed, looks like a duplicate of ros2/rmw_opensplice#264.

closed 2019-05-16 by jacobperron