ros2/demos #341: action_tutorials Python Fibonacci Client/Server not executable

clalancette, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 0.7.4 (0b54af7f83dd3aca000d04c6b69a2c25d1c09803)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
ros2 run action_tutorials fibonacci_action_server.py
```
or
```
ros2 run action_tutorials fibonacci_action_client.py
```

#### Expected behavior

Running action server or client.

#### Actual behavior

```
Traceback (most recent call last):
  File "/home/ubuntu/ros2_ws/install/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.7.0', 'console_scripts', 'ros2')()
  File "/home/ubuntu/ros2_ws/install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/ubuntu/ros2_ws/install/ros2run/lib/python3.6/site-packages/ros2run/command/run.py", line 72, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "/home/ubuntu/ros2_ws/install/ros2run/lib/python3.6/site-packages/ros2run/api/__init__.py", line 61, in run_executable
    process = subprocess.Popen(cmd)
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/ubuntu/ros2_ws/install/action_tutorials/lib/action_tutorials/fibonacci_action_server.py'
```

#### Additional information
Since this is called out in our Python action client [tutorial](https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Client-Python/), we should probably fix this before Dashing.

jacobperron, 2019-05-22:
Looks like the scripts are missing shebang lines.

clalancette, 2019-05-22:
Shouldn't this be done with with the entry_points, though, similar to https://github.com/ros2/demos/blob/master/demo_nodes_py/setup.py#L34 ?

jacobperron, 2019-05-22:
I'm not sure. I didn't see many examples of adding a `setup.py` alongside a CMake package. I can give it a try to see if it works.

PR #343 (2019-05-22): Add missing shebang to action tutorial executables
Fixes #341 

I briefly looked into using a setup.py instead as suggested in https://github.com/ros2/demos/issues/341#issuecomment-494851271, but I'm not sure how to do this in an ament CMake package. The only example I found was [message_filters](https://github.com/ros2/message_filters/), which is a Cmake package with a setup.py, but it doesn't appear to make use of the info in setup.py AFAICT.
closed 2019-05-22 by jacobperron