ros2/demos #344: External logger level configuration unable to be set in logging_demo

skucheria, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Debian packages
- Version or commit hash:
  - 0.7.3
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
``` ros2 run logging_demo logging_demo_main ```

In another terminal:
``` ros2 service call /config_logger logging_demo/ConfigLogger "{logger_name: 'logger_usage_demo', level: INFO}" ```

#### Expected behavior
The service call will set the level of the demo’s logger back to ```INFO``` with output:

``` waiting for service to become available...
     requester: making request: 
     logging_demo.srv.ConfigLogger_Request(logger_name='logger_usage_demo', level='INFO')

     response:
     logging_demo.srv.ConfigLogger_Response(success=True) ```

#### Actual behavior
The service is unable to run and returns the following error:

``` Traceback (most recent call last):
  File "/opt/ros/dashing/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.7.0', 'console_scripts', 'ros2')()
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2service/command/service.py", line 43, in main
    return extension.main(args=args)
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2service/verb/call.py", line 56, in main
    args.service_type, args.service_name, args.values, period)
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2service/verb/call.py", line 66, in requester
    srv_module = getattr(module, srv_name)
AttributeError: module 'logging_demo' has no attribute 'ConfigLogger' ```

dirk-thomas, 2019-05-22:
Currently for this to work you need to pass `logging_demo/srv/ConfigLogger`:

```
ros2 service call /config_logger logging_demo/srv/ConfigLogger "{logger_name: 'logger_usage_demo', level: INFO}"
```

ros2/ros2_documentation#213 updates the documentation.

Independent of that I will work on a patch which keeps the old style working.

dirk-thomas, 2019-05-22:
> Independent of that I will work on a patch which keeps the old style working.

See ros2/ros2cli#247.

closed 2019-05-22 by dirk-thomas