ros2/demos #346: Alternate node name for component_container results in warning

clalancette, 2019-05-22:
## Bug report

I'm following the tutorial at https://index.ros.org/doc/ros2/Tutorials/Composition/#remapping-container-name-and-namespace .  When I run the component_container with an alternate node name (by specifying `__node:=MyContainer`), I always get a warning when inserting components into the container.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 45c8fa990457f2419e00bcc33dd5b3a8a9a6a681
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp_components

#### Steps to reproduce issue

In terminal 1:
```
ros2 run rclcpp_components component_container __node:=MyContainer
```

In terminal 2:
```
ros2 component load /MyContainer composition composition::Listener
```

#### Expected behavior
Composition Listener is loaded into the container without warnings.

#### Actual behavior
In the terminal running the component_container:
```
[INFO] [MyContainer]: Load Library: /home/ubuntu/ros2_ws/install/composition/lib/liblistener_component.so
[INFO] [MyContainer]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [MyContainer]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

#### Additional information
If I run the component_container without `__node:=MyContainer`, no error is shown.  Also, using a namespace (`__ns:=/ns`) doesn't cause this to happen, only the change in node name.

cottsay, 2019-05-22:
I think that this is fundamentally the same issue as ros2/launch#204

dirk-thomas, 2019-05-23:
> I think that this is fundamentally the same issue as ros2/launch#204

The referenced issue is specific to a single executable which contains multiple nodes. There is currently no way to pass node-specific arguments in that case and all arguments are used by all nodes.

This case is different where the manager (without any components) is started and can optionally use arguments. The later started components should not be subject to these arguments though (which they currently are, hence the warning).

See ros2/rclcpp#736 for the proposed fix.

closed 2019-05-23 by dirk-thomas