ros2/demos #347: Lifecycle service client doesn't quit on Ctrl-C

clalancette, 2019-05-22:
## Bug report

I'm following the tutorial at https://index.ros.org/doc/ros2/Tutorials/Managed-Nodes/ .  All stages of the tutorial seem to work well until I go to shut it down.  When I press Ctrl-C to kill the command `ros2 run lifecycle lifecycle_service_client`, it hangs, seemingly forever.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 45c8fa990457f2419e00bcc33dd5b3a8a9a6a681
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Three different terminals:

Terminal 1:
```
ros2 run lifecycle lifecycle_talker
```

Terminal 2:
```
ros2 run lifecycle lifecycle_listener
```

Terminal 3:
```
ros2 run lifecycle lifecycle_service_client
```

Then Ctrl-C the talker, followed by the listener, followed by the client.

#### Expected behavior

The demo runs, then all nodes can be killed by hitting Ctrl-C.

#### Actual behavior

The talker and listener nodes are killed as expected, by the service_client lingers forever.

#### Additional information

If I just start the `service_client` by itself (without the talker or listener), I can Ctrl-C it just fine.  It's only after it has caused some transitions to happen that it hangs.

clalancette, 2019-05-22:
(a similar thing happens when using `ros2 launch lifecycle lifecycle_demo.launch.py` to launch the demo, but there the launch file eventually escalates to SIGTERM to kill it off)

dirk-thomas, 2019-05-22:
Can you clarify in which moment you are Ctrl-C-ing the processes? I tried it right after getting into `inactive` as well as `active` and in both cases the client exists just fine with a last error message:

> [ERROR] [lc_client]: Service /lc_talker/change_state is not available.

clalancette, 2019-05-22:
> Can you clarify in which moment you are Ctrl-C-ing the processes? I tried it right after getting into `inactive` as well as `active` and in both cases the client exists just fine with a last error message:

So it turns out I wasn't waiting long enough.  Indeed, if I Ctrl-C right after transitioning to "inactive", then wait 8-10 seconds, it does quit.  The same goes after the transition to "active".  If I wait until it goes all the way to "finalized", then a Ctrl-C is instantaneous.

So it is not that it won't quit, just that it is taking long enough that I got impatient.  Also, `launch` gets impatient as well.  I'll update the title to reflect that.

PR #348 (2019-05-22): use rate instead of thread::sleep to react to Ctrl-C
Fixes #347 by using a `WallRate.sleep` instead of `std::this_thread::sleep_for`.

The additional check of `rclcpp::ok` after the sleep ensure to not print an error message when Ctrl-C-ing the process:

> [ERROR] [lc_client]: Service /lc_talker/change_state is not available.
closed 2019-05-22 by dirk-thomas