ros2/demos #351: Liveliness demos with Fast-RTPS don't seem to work

clalancette, 2019-05-23:
## Bug report

I'm following the Quality of Service demos at https://github.com/ros2/demos/blob/master/quality_of_service_demo/README.md , and none of the liveliness demos seem to work for me.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - cee4332944d9250ecf61b8492b71aa65a40ff972
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp/rclpy

#### Steps to reproduce issue
```
ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
```

#### Expected behavior
After 2 seconds, the publisher is killed and the subscriber gets a callback which prints out some information about liveliness.

#### Actual behavior
After 2 seconds, the publisher seems to be killed (at least, it stops printing), but we get no information about liveliness.

#### Additional information
This is failing for both `quality_of_service_demo_cpp` and `quality_of_service_demo_py`.  Further, trying to run `ros2 run quality_of_service_demo_cpp liveliness 250 --node-assert-period 0 --policy MANUAL_BY_NODE` throws an exception on startup saying "Manual liveliness unsupported for fastrtps", but that is not mentioned in the README.  @emersonknapp FYI.

emersonknapp, 2019-05-23:
I have opened this PR to disable all Liveliness for Fast-RTPS https://github.com/ros2/rmw_fastrtps/pull/282 - we thought Liveliness would be supported in the 1.8 release but it ended up not being the case.

Do you think we should also update the demo README to mention this limitation?

closed 2019-05-23 by dirk-thomas

jacobperron, 2019-05-30:
I would update the README since it might be perceived as a bug to users (https://github.com/ros2/demos/issues/365).