ros2/demos #353: dummy_robot_bringup references non-existent rrbot package

clalancette, 2019-05-23:
## Bug report

When trying to visualize the robot from the dummy_robot_bringup launch file, the Rviz2 RobotModel plugin complains that it can't find the rrbot_description package.  As far as I can tell, that package doesn't exist anywhere.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - cee4332944d9250ecf61b8492b71aa65a40ff972
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Terminal 1:
```
ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
```

Terminal 2:
```
rviz2 &
```

#### Expected behavior
Rviz2 can visualize the dummy robot without issue.

#### Actual behavior
RViz2 throws a bunch of errors, and displays white boxes on the screen:
```
[ERROR] [rviz2]: Could not load resource [package://rrbot_description/model/meshes/hokuyo.dae]: 
[ERROR] [rviz2]: FileNotFoundException: Cannot locate resource package://rrbot_description/model/meshes/hokuyo.dae in resource group OgreAutodetect. in ResourceGroupManager::openResource at /home/ubuntu/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreResourceGroupManager.cpp (line 708)
[ERROR] [rviz2]: could not load model 'package://rrbot_description/model/meshes/hokuyo.dae' for link 'single_rrbot_hokuyo_link': FileNotFoundException: Cannot locate resource package://rrbot_description/model/meshes/hokuyo.dae in resource group OgreAutodetect. in ResourceGroupManager::openResource at /home/ubuntu/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreResourceGroupManager.cpp (line 708)
[ERROR] [rviz2]: Could not load resource [package://rrbot_description/model/meshes/hokuyo.dae]: 
[ERROR] [rviz2]: FileNotFoundException: Cannot locate resource package://rrbot_description/model/meshes/hokuyo.dae in resource group OgreAutodetect. in ResourceGroupManager::openResource at /home/ubuntu/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreResourceGroupManager.cpp (line 708)
[ERROR] [rviz2]: could not load model 'package://rrbot_description/model/meshes/hokuyo.dae' for link 'single_rrbot_hokuyo_link': FileNotFoundException: Cannot locate resource package://rrbot_description/model/meshes/hokuyo.dae in resource group OgreAutodetect. in ResourceGroupManager::openResource at /home/ubuntu/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreResourceGroupManager.cpp (line 708)
```

#### Additional information

Karsten1987, 2019-05-24:
There is already a PR open for this: https://github.com/ros2/demos/pull/349

Karsten1987, 2019-05-29:
fixed by #349

closed 2019-05-29 by Karsten1987