ros2/demos #354: composition launch won't terminate after first Ctrl-C

hidmic, 2019-05-24:
## Bug report

**Required Info:**

- Operating System:
  - Windows (Server) 10
- Installation type:
  - From binary fat archive.
- Version or commit hash:
  - [Dashing pre-release](https://ci.ros2.org/view/packaging/job/packaging_windows/1382/)
- DDS implementation:
  - eProsima Fast-RTPS, RTI Connext

#### Steps to reproduce issue

Run the composition demo launch:

```sh
ros2 launch composition composition_demo.launch.py
```

After initial setup, attempt to Ctrl-C it.

#### Expected behavior

Launched processes are interrupted and the launch CLI itself eventually exits gracefully.

#### Actual behavior

Launched processes are interrupted but the launch CLI won't exit. After a second Ctrl-C, output reads:

```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\dev\ros2-windows\Lib\site-packages\launch\launch_service.py", line 62, in close_loop
    loop.close()
  File "c:\python37\lib\asyncio\proactor_events.py", line 537, in close
    self._proactor.close()
  File "c:\python37\lib\asyncio\windows_events.py", line 803, in close
    self._poll(msg_update)
  File "c:\python37\lib\asyncio\windows_events.py", line 715, in _poll
    status = _overlapped.GetQueuedCompletionStatus(self._iocp, ms)
  File "C:\dev\ros2-windows\Lib\site-packages\launch\utilities\signal_management.py", line 119, in __on_sigint
    __original_sigint_handler(signum, frame)
KeyboardInterrupt
```

ivanpauno, 2019-05-24:
Same happens when testing from source.

ivanpauno, 2019-05-28:
Using `rmw_connext_cpp`, it happens too with `manual composition` and `dlopen composition` demos.
Steps to reproduce in the [tutorial](https://index.ros.org/doc/ros2/Tutorials/Composition/).

Info: Windows server 2019 datacenter, from source install, current master.

ivanpauno, 2019-05-30:
> Using `rmw_connext_cpp`, it happens too with `manual composition` and `dlopen composition` demos.
> Steps to reproduce in the [tutorial](https://index.ros.org/doc/ros2/Tutorials/Composition/).
> 
> Info: Windows server 2019 datacenter, from source install, current master.

Ok, this is finally not the same. In general, when  using connext, it takes time too close after a `cntrl+c`. But if you wait, it actually closes (not double cntrl+c needed).

closed 2019-08-06 by ivanpauno