ros2/demos #362: Intra_process_demo camera_node and watermark_node print tons of warnings

clalancette, 2019-05-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 8071da18e06e15cddc639360f66ea31022d16ee4
- DDS implementation:
  - Opensplice 6.9
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Terminal 1:
```
ros2 run intra_process_demo camera_node
```

Terminal 2:
```
ros2 run intra_process_demo watermark_node
```

#### Expected behavior

Both nodes run, passing data between them.

#### Actual behavior

Both nodes run, and data gets passed, but `camera_node` prints once:
```
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
```

And `watermark_node` continually prints:
```
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
[WARN] [rclcpp]: Intra process message no longer being stored when trying to handle it
```

#### Additional information

I don't see this with `image_pipeline_all_in_one` or with `image_pipeline_with_two_image_view`

dirk-thomas, 2019-05-30:
Are you working on this? If yes, please assign it to yourself and add it to the project board.

clalancette, 2019-05-30:
> Are you working on this? If yes, please assign it to yourself and add it to the project board.

Not at the moment, no.  I have a couple of other things that need to get done first, then I may have time to come back to this.  In the meantime, if someone else wants to pick it up, that would be fine.

closed 2019-05-30 by ivanpauno