ros2/demos #366: Document that Connext-6.0.0 does not

prajakta-gokhale, 2019-05-31:
QoS tutorials don't work with `Connext 6.0.0`.
## Bug report

**Required Info:**

- Operating System:
  - macOS Mojave
- Installation type:
  - binaries 
- Version or commit hash:
  - 
- DDS implementation:
  - Connext 
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue

```
➜ export NDDSHOME=/Applications/rti_connext_dds-6.0.0
➜ export RMW_IMPLEMENTATION=rmw_connext_cpp

➜ ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
lifespan(75025,0x10d3bc5c0) malloc: *** error for object 0x7344: pointer being freed was not allocated
lifespan(75025,0x10d3bc5c0) malloc: *** set a breakpoint in malloc_error_break to debug

➜ ros2 run quality_of_service_demo_cpp liveliness 250 --node-assert-period 0 --policy MANUAL_BY_NODE
liveliness(75039,0x10c8525c0) malloc: *** error for object 0x7344: pointer being freed was not allocated
liveliness(75039,0x10c8525c0) malloc: *** set a breakpoint in malloc_error_break to debug

➜ ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
deadline(75049,0x10d8f15c0) malloc: *** error for object 0x7344: pointer being freed was not allocated
deadline(75049,0x10d8f15c0) malloc: *** set a breakpoint in malloc_error_break to debug
```

#### Expected behavior
QoS tutorials should work with Connext 6.0.0

#### Actual behavior
QoS tutorials crash with Connext 6.0.0

ivanpauno, 2019-05-31:
I think we aren't officially supporting RTI Connext 6.0.0. Only 5.3.1 at the moment.
Check `RMW DDS Milddleware Providers` table in `Dashing Diademata (May 2019 - May 2021)` subsection [here](https://github.com/ros-infrastructure/rep/blob/master/rep-2000.rst).

dirk-thomas, 2019-05-31:
Closing as duplicate of ros2/ros2_documentation#257.

closed 2019-05-31 by dirk-thomas

jacobperron, 2019-05-31:
I don't know if this is exactly a duplicate of ros2/ros2_documentation#257. This ticket is about supporting Connext 6.0.0, compared to updating documentation related to 5.3.1. If supporting Connext 6.0.0 is something we plan to do in the future, then I think this ticket (or an equivalent) should be open for visibility.

dirk-thomas, 2019-05-31:
The PR addressing the referenced issue (https://github.com/ros2/ros2_documentation/pull/259/) mentions that 5.3 is supported and 6 is currently not.

Also this is nothing which should be done in this repo.