ros2/demos #368: shared library error on lifecycle demos

jasonbeach, 2019-06-08:
I just upgraded from crystal to dashing.  I've updated my local copy of this repo and switched to the dashing branch. Repo is cloned in `~/ros2_example_ws/src/demos` 
```
cd ~/ros2_example_ws
colcon build
source install/setup.bash
ros2 run lifecycle lifecycle_talker
```
I get this error

```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialized rcl init options: failed to load shared library of rmw implementation. Exception: Cannot load library: /opt/ros/dashing/lib/librmw_fastrtps_cpp.so: undefined symbol: _ZTVN8eprosima8fastrtps28DisablePositiveACKsQosPolicyE, at /tmp/binarydeb/ros-dashing-rmw-implementation-0.7.1/src/functions.cpp:137, at /tmp/binarydeb/ros-dashing-rcl-0.7.4/src/rcl/init_options.c:55
```
I'm running on Ubuntu 18.04 and have only installed the Fast-RTPS that comes from `sudo apt install ros-dashing-desktop`

Any ideas what it'd be?

dirk-thomas, 2019-06-08:
Likely you are using incompatible versions of packages. Please double check which versions of binaries you have download and what you use from source.

In order for us to help we would need exact steps to reproduce. Probably when putting together such steps the difference will become obvious.

jasonbeach, 2019-06-08:
Hi--thanks for responding so quickly.  Here are some excerpts from my `.bash_history` (which unfortunately was a little jumbled / incomplete because of using multiple terminals and tmux)

Starting from `~/ros2_example_ws/src/demos` which is my local copy of this repo
```
git pull
git checkout dashing
sudo apt remove ros-crystal-desktop 
sudo apt install ros-dashing-desktop # this failed because it couldn't find the package
sudo apt update 
```
This failed when hitting the ros server saying there was an error with the key (I don't remember exactly what the error was). I re-added the key which I had done when installing crystal, but for whatever reason it apparently needed to be done again:
```
curl http://repo.ros2.org/repos.key | sudo apt-key add -
less /etc/apt/sources.list.d/ros2-latest.list 
```
this file contains 
`deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu bionic main`
Continuing on,
```
sudo apt update
sudo apt install ros-dashing-desktop
sudo apt autoremove # was trying to remove crystal packages
cd ~/ros2_example_ws/
ls
rm -rf build/ install/ log/
colcon build
```
At this point I had a lot of python errors.  It appeared they stemmed from me having installed Anaconda which I had installed in between installing crystal and doing this update.  I solved this by commenting out the conda init statements in my .bashrc file so that the system python was used, however build still failed, this time because I was missing some cmake files related to colcon / ament.  I solved this by:
```
sudo apt install ros-dashing-launch-testing-ament-cmake 
sudo apt install ros-dashing-ros-testing 
```

at which point the build succeeded. I then tried to run them via

```
source install/setup.bash 
ros2 run  lifecycle lifecycle_talker 
```
and got the error above.  The issue is not unique to the lifecycle demo.  This fails with the same error:
```
ros2 run intra_process_demo two_node_pipeline 
```
I agree it sounds very ABI-related, I just haven't had to time to really look into it.  At some point I can try installing ros2 from source, but in the mean time any help would be appreciated.

jasonbeach, 2019-06-16:
So I completely nuked my ros2 install. 
```
sudo apt purge ros-dashing-desktop 
sudo apt autoremove
sudo rm -rf /opt/ros/dashing/
sudo apt purge ros-crystal-*
sudo apt purge ros-dashing-*
sudo rm -rf /opt/ros/crystal/
sudo rm -rf /opt/ros/
```
and then reinstalled it via 
`sudo apt install ros-dashing-deskop`

to get the examples to build, I had to also install these packages:
```
sudo apt install ros-dashing-launch-testing-ament-cmake 
sudo apt install ros-dashing-ros-testing 
```
I also had to do a work around for Anaconda to ensure safe search paths given here: https://github.com/pism/pism/issues/356.  So the last few lines of my ~/.bashrc file are
```
source /opt/ros/dashing/setup.bash
export NOCONDA_PATH=$PATH

# >>> conda initialize >>>
# !! Contents within this block are managed by 'conda init' !!
__conda_setup="$('/home/jason/anaconda3/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
    if [ -f "/home/jason/anaconda3/etc/profile.d/conda.sh" ]; then
        . "/home/jason/anaconda3/etc/profile.d/conda.sh"
    else
        export PATH="/home/jason/anaconda3/bin:$PATH"
    fi
fi
unset __conda_setup
# <<< conda initialize <<<
```
Specifically I had to add the line `export NOCONDA_PATH=$PATH` and then to build the examples
```
cd ros2_example_ws/
PATH=$NOCONDA_PATH colcon build --symlink-install
```
and so far I've been able to run the demos fine.

jacobperron, 2019-06-19:
@jasonbeach Correct me if I'm wrong, but it sounds like you've resolved the issue. If you experience the issue again, feel free to comment and we can re-open the issue.

closed 2019-06-19 by jacobperron

Sensenzhl, 2022-01-02:
I also meet this problem today.
Has it been solved? Is there any solution available?
Thanks.

jacobperron, 2022-01-05:
@Sensenzhl This is a very old issue. For visibility, it might be best to open a new ticket with details about the version of ROS you are using and detailed steps on how someone else could reproduce.