ros2/demos #374: Install python programs properly

ivanpauno, 2019-06-28:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - from source
- Version or commit hash:
  - current master https://github.com/ros2/launch_ros/commit/37929eea57f2c7863207a9f808d8b16b73464de9 f78dcc981cc783aad162e2a8530a7aa23c4fda21
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

The following launch file:
```python
import launch
from launch_ros.actions import Node

def generate_launch_description():
    """Generate launch description with multiple components."""
    return launch.LaunchDescription([
        Node(
            package='action_tutorials',
            node_executable='fibonacci_action_server.py',
        )
    ])
```

#### Expected behavior
The fibonacci action server should be launched.

#### Actual behavior
The node crashes when launching the `Node` executable, because the installed file is not a proper program on Windows.

#### Additional information

I don't know if we have some cmake sugar for correctly installing python executables in an `ament_cmake` package.

In `ament_python` packages, we are using `console_scripts` `entry_points`, which work properly:
https://github.com/ros2/demos/blob/f78dcc981cc783aad162e2a8530a7aa23c4fda21/demo_nodes_py/setup.py#L34-L45
But in `ament_cmake` packages, the usual install macro doesn't produce the expected result:
https://github.com/ros2/demos/blob/f78dcc981cc783aad162e2a8530a7aa23c4fda21/action_tutorials/CMakeLists.txt#L21-L26

sloretz, 2019-06-29:
IIUC the `console_scripts` entry point creates a wrapper executable that launches python and imports/calls a function given. It looks like the install rule above is just copying a python script. I don't think `lib/${PROJECT_NAME}` is the a good spot for plain python scripts since they won't work with `ros2 run` in the general case.

At the very least the launch file should invoke python with the path to the scripts (not sure where, maybe in `share`?) instead. Alternatively, python code could be put in a python package that `action_tutorials` depends on.

ivanpauno, 2019-07-01:
> IIUC the `console_scripts` entry point creates a wrapper executable that launches python and imports/calls a function given. It looks like the install rule above is just copying a python script.

Yes, it's just copying it. My idea was to add a cmake macro in `ament_cmake_python`, which will work similarly to `console_scripts entry_points`.

> I don't think `lib/${PROJECT_NAME}` is the a good spot for plain python scripts since they won't work with `ros2 run` in the general case.

I also think it's not a good idea to install plain python scripts there. `ros2 run` works, because it does a little trick:

https://github.com/ros2/ros2cli/blob/5a3c2f1543b2814148cc06e60813e805ca196f48/ros2run/ros2run/api/__init__.py#L51-L56

> At the very least the launch file should invoke python with the path to the scripts (not sure where, maybe in `share`?) instead. Alternatively, python code could be put in a python package that `action_tutorials` depends on.

I think that `action_tutorials` can be switched to a `ament_python` package. We already have the `Fibonacci` action definition in `example_interfaces`, we could use that one.

https://github.com/ros2/example_interfaces/blob/master/action/Fibonacci.action

This is probably happening in other packages too. I think that it would be good to have the ability to install a `python` program from an `ament_cmake` package correctly.

jacobperron, 2019-07-01:
>  think that action_tutorials can be switched to a ament_python package. We already have the Fibonacci action definition in example_interfaces, we could use that one.

I purposefully made `action_tutorials` a cmake package since I'd planned to add the C++ tutorial code to the same package. We also build the Fibonacci action as part of the [Creating an Action](https://index.ros.org/doc/ros2/Tutorials/Actions/Creating-an-Action/) tutorial. IMO it's nice to keep all of the action tutorial code in one repo (and possible migrate the tutorials pages here too). Perhaps it's worth considering putting the Python and C++ tutorials into separate packages, but I'm not sure if it's necessary.

hidmic, 2019-07-01:
IMHO there's no reason to switch the `action_tutorials` package to a pure Python package, even more so considering @jacobperron's point. And to that end, I agree with @ivanpauno in that `ament_cmake` should install Python scripts properly on Windows, maybe a new `ament_cmake_python` macro?

jacobperron, 2019-07-01:
So, `rclpy` is a cmake package of Python code. It calls [`ament_python_install_package`](https://github.com/ros2/rclpy/blob/95a09a0e66a05e0c1fe7e5839d0a5ae99ca73c0a/rclpy/CMakeLists.txt#L34); maybe that's what we want?

ivanpauno, 2019-07-01:
> So, `rclpy` is a cmake package of Python code. It calls [`ament_python_install_package`](https://github.com/ros2/rclpy/blob/95a09a0e66a05e0c1fe7e5839d0a5ae99ca73c0a/rclpy/CMakeLists.txt#L34); maybe that's what we want?

I think that's only for installing a python package, but it doesn't have a way of specifying the `console_scripts` as it doesn't use a `setup.py`.

jacobperron, 2019-07-01:
Right, that's why I didn't use it..

Maybe we could add support to `ament_python_install_package` for providing optional entry points or a new cmake function as you suggest. I'm not sure what makes the most sense.

nuclearsandwich, 2019-12-05:
@jacobperron @ivanpauno I think that this was resolved in mainline by splitting the action_tutorials package into separate packages. I think this is a problem that should be fixed in Dashing but I'm not sure if backporting the split packages is the way to go, although it would probably be the simplest.

Alternatives I can come up with are simulating the console_scripts/entrypoints behavior by installing wrapper scripts on Windows or backporting the new packages and leaving this one as a metapackage (although that wouldn't allow us to call the executables / launch files from this package so it is still incomplete)

jacobperron, 2019-12-05:
> I think that this was resolved in mainline by splitting the action_tutorials package into separate packages.

Correct.

Backporting the split packages would be okay I think. Though it doesn't really *fix* the existing `action_tutorials` package, instead  it provides alternatives. Maybe simulating the console_scripts behavior on Windows is the way to go, but I'm not sure how involved doing this is.

jacobperron, 2019-12-05:
BTW, we could probably close this issue since it's resolved for Eloquent+.

> My idea was to add a cmake macro in ament_cmake_python, which will work similarly to console_scripts entry_points.

@ivanpauno, if you think this would still be useful, maybe it's worth opening a ticket for it (and if such a feature could be backported that would also help resolve this issue for Dashing.

ivanpauno, 2019-12-10:
> @ivanpauno, if you think this would still be useful, maybe it's worth opening a ticket for it (and if such a feature could be backported that would also help resolve this issue for Dashing).

Yeah, I think it will be useful, specially for backporting.
In general, I would avoid installing python programs from an `ament_cmake` package.
Opened an issue there https://github.com/ament/ament_cmake/issues/213.

I'm closing this one (but I will left it in the Dashing board).

closed 2019-12-10 by ivanpauno

nuclearsandwich, 2019-12-10:
Since there is no concrete fix I am bumping the proposed change to a future patch release.