ros2/demos #407: Process hung on [INFO] [rclcpp]: signal_handler(signal_value=2)

claireyywang, 2019-10-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Installation type:
  - Source 
- Version or commit hash:
  - Eloquent 
- DDS implementation:
  - Connext 
- Client library (if applicable):
  - rclcpp 

#### Steps to reproduce issue

```
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 run quality_of_service_demo_cpp interactive_publisher --delay 0.5 --deadline 1 --liveliness MANUAL_BY_TOPIC --lease 1
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 run quality_of_service_demo_cpp interactive_subscriber --deadline 1 --liveliness MANUAL_BY_TOPIC --lease 1
```

#### Expected behavior
print
```
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
```
then exit the process
#### Actual behavior
```
[INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [rclcpp]: signal_handler(signal_value=2)
```
never exit.

clalancette, 2019-10-23:
Out of curiousity, how long did you wait?  We have a known issue somewhere where Connext can take a substantial amount of time to cleanup after itself (like 10 seconds or something).  I'm wondering if this is a duplicate.

claireyywang, 2019-10-23:
> Out of curiousity, how long did you wait? We have a known issue somewhere where Connext can take a substantial amount of time to cleanup after itself (like 10 seconds or something). I'm wondering if this is a duplicate.

I tried again, waited for at least five minutes and got the same result. I had to close the terminal to kill it.

clalancette, 2019-10-23:
> > 
> 
> I tried again, waited for at least five minutes and got the same result. I had to close the terminal to kill it.

Sounds like something new :+1:

hidmic, 2019-11-13:
I came across the same thing. @claireyywang try pressing `x`...

The thing is that these demos spin up a separate thread to support `stdin`-based user interaction. Because of the way that bit is coded, you have to either Ctrl-C or press `x` (to stop the executor) and then press another random key to get yourself out of a blocking `getch*()` call. 

I fiddled a bit on how to fix it but being cross-platform is going to be time consuming. I'll come back later.

wjwwood, 2019-12-12:
@claireyywang can you push on this issue. @hidmic says the code may need to be refactored to close when doing ctrl-c. We can either backlog this or fix it. Thanks!

claireyywang, 2019-12-13:
@wjwwood  sure I'll look into it

clalancette, 2024-07-26:
Since we don't support `rmw_connext_cpp` anymore, I'm going to go ahead and close this out.

closed 2024-07-26 by clalancette