ros2/demos #412: Intra_process_demo is not zero-copy if doing external communications as well

AlexisTM, 2019-10-30:
## Bug report

Intra process communication has a 

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binairies
- Version or commit hash:
  - [c9e55acba6eaf377ccacce844853fd9eb20d1113](https://github.com/ros2/demos/tree/c9e55acba6eaf377ccacce844853fd9eb20d1113)
- DDS implementation:
  - Fast-RTPS
- Client library:
  - rclcpp

#### Steps to reproduce issue

* Run `ros2 run intra_process_demo two_node_pipeline`
* Experience zero-copy (received object address == published object address)
* Run `ros2 topic echo number`
* The received object doesn't have the same address as the published one anymore

#### Expected behavior

I expect the received object address to always be the published object address even when also subscribing to the topic externally.

#### Actual behavior

The received object has been copied.

#### Additional information

> NOTE: The object is reallocated to the same address in the lambda function

> NOTE: The Published address when listening is equal to the last received address; Due to allocation optimization?

```bash
$ ros2 run intra_process_demo two_node_pipeline
Published message with value: 0, and address: 0x563E68561220
 Received message with value: 0, and address: 0x563E68561220
Published message with value: 1, and address: 0x563E68561220
 Received message with value: 1, and address: 0x563E68561220
Published message with value: 2, and address: 0x563E68561220
 Received message with value: 2, and address: 0x563E68561220
Published message with value: 3, and address: 0x563E68561220
 Received message with value: 3, and address: 0x563E68561220
Published message with value: 4, and address: 0x563E68561220
 Received message with value: 4, and address: 0x563E68561220
Published message with value: 5, and address: 0x563E68561220
 Received message with value: 5, and address: 0x563E68561220
Published message with value: 6, and address: 0x563E68561220
 Received message with value: 6, and address: 0x563E68561220
Published message with value: 7, and address: 0x563E68561220
 Received message with value: 7, and address: 0x563E68561220
Published message with value: 8, and address: 0x563E68561220
 Received message with value: 8, and address: 0x563E685D29D0
Published message with value: 9, and address: 0x563E685D29D0
 Received message with value: 9, and address: 0x563E68561220
Published message with value: 10, and address: 0x563E68561220
 Received message with value: 10, and address: 0x563E685D29D0
Published message with value: 11, and address: 0x563E685D29D0
 Received message with value: 11, and address: 0x563E68561220
Published message with value: 12, and address: 0x563E68561220
 Received message with value: 12, and address: 0x563E68561220
```

```bash
$ ros2 topic echo number
data: 8
---
data: 9
---
data: 10
---
data: 11
---
```

hidmic, 2019-11-04:
@AlexisTM what `rclcpp` version are you running against?

AlexisTM, 2019-11-05:
`ros-dashing-rclcpp/bionic,now 0.7.11-1bionic.20191016.192455 amd64 [installed,automatic]`

hidmic, 2019-11-05:
Alright, yeah, I see the problem. I'll open an issue on `rclcpp`, to assess whether to hotfix or backport (as this is fixed in Eloquent). Thanks for reporting!

AlexisTM, 2019-11-05:
Closing in favor of: https://github.com/ros2/rclcpp/issues/912

closed 2019-11-05 by AlexisTM