ros2/demos #417: [action_tutorials_py] fibonacci_action_client fails in an unclear way when more than one action server is available

ivanpauno, 2019-11-18:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  from source
- Version or commit hash:
  -  2793f36345f94fb8e86dc5bf2ab9c6574fa7c14c
- DDS implementation:
  -  Fast-RTPS, OpenSplice, Connext, CycloneDDS (I did test all of them)
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

Terminal 1:
```
ros2 run action_tutorials_py fibonacci_action_server
```

Terminal 2:
```
ros2 run action_tutorials_py fibonacci_action_server
```

Terminal 3:
```
ros2 run action_tutorials_py fibonacci_action_client
```

#### Expected behavior

The action client should succeed, or fail with an error clearly indicating that the problem is that there are two action servers available.

#### Actual behavior

fibonacci_action_client output:
```bash
[INFO] [fibonacci_action_client]: Goal accepted :)
Traceback (most recent call last):
  File "/home/ivanpauno/ros2_ws/install/action_tutorials_py/lib/action_tutorials_py/fibonacci_action_client", line 11, in <module>
    load_entry_point('action-tutorials-py', 'console_scripts', 'fibonacci_action_client')()
  File "/home/ivanpauno/ros2_ws/build/action_tutorials_py/action_tutorials_py/fibonacci_action_client.py", line 71, in main
    rclpy.spin(action_client)
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/__init__.py", line 190, in spin
    executor.spin_once()
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 684, in spin_once
    raise handler.exception()
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/task.py", line 210, in __call__
    self._handler.send(None)
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 404, in handler
    await call_coroutine(entity, arg)
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 370, in _execute_waitable
    await waitable.execute(data)
  File "/home/ivanpauno/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/action/client.py", line 283, in execute
    self._sequence_number_to_goal_id[sequence_number],
KeyError: 1
```

Both action servers accept the goal and execute it correctly.

#### Additional information

It would be interesting to test the same thing with `action_tutorials_cpp`.

ivanpauno, 2019-11-18:
@jacobperron you may have a better idea than me of what's going on here.

sloretz, 2019-11-18:
It might be that having two action servers (or two service servers) has to be undefined. The client could call [`get_service_names_and_types_by_node`](https://github.com/ros2/rclpy/blob/da268097d74800e8fdf0bf539296f5b843b93c17/rclpy/rclpy/node.py#L1525) and look for duplicate `send_goal` services, but a second server could be discovered between checking and sending the goal or requesting the result.

jacobperron, 2019-12-06:
I agree with @sloretz, that having two or more action servers (or service servers) with the same name is undefined behavior. Even if we detect multiple services with the same name, how can we limit the call to one of them (and should we)?

In any case, we certainly should avoid crashing. I tried doing the same thing with `action_tutorials_cpp` and at least it doesn't crash. After both action servers reply, we see the client claims to ignore the second goal response, but there is double feedback:

<details>
<summary>output</summary>
<pre>
$ ros2 run action_tutorials_cpp fibonacci_action_client
[INFO] [fibonacci_action_client]: Sending goal
[INFO] [fibonacci_action_client]: Goal accepted by server, waiting for result
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 
[ERROR] [fibonacci_action_client.rclcpp_acton]: unknown goal response, ignoring...
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 55 
[INFO] [fibonacci_action_client]: Next number in sequence received: 0 1 1 2 3 5 8 13 21 34 55 
[INFO] [fibonacci_action_client]: Result received: 0 1 1 2 3 5 8 13 21 34 55
</pre>
</details>

Maybe we create a ticket to discuss how to handle the case of multiple action servers with the same name, or add a note to the design doc that this case is undefined behavior.
In the meantime, I'll look at resolving the immediate issue of the action client crashing in rclpy.

jacobperron, 2019-12-06:
Fixed in https://github.com/ros2/rclpy/pull/474

closed 2019-12-07 by jacobperron