ros2/demos #418: error when compile the `demo_nodes_cpp`

friendsocket, 2019-11-26:
When i compiled the **demo_nodes_cpp**, it has an error occur.
```
$ cd ~/demos/demo_nodes_cpp
$ source ~/ros2_ws/install/local_setup.bash
$ mkdir build && cd build
$ cmake ..
```
then the compile error occur.
```
....
-- Found rclcpp_components: 0.7.7 (/home/dbt/ros2_ws/install/rclcpp_components/share/rclcpp_components/cmake)
-- Found poco_vendor:  (/home/dbt/ros2_ws/install/poco_vendor/share/poco_vendor/cmake)
-- Found std_msgs: 0.7.0 (/home/dbt/ros2_ws/install/std_msgs/share/std_msgs/cmake)
CMake Error at CMakeLists.txt:81 (rclcpp_components_register_node):
  Unknown CMake command "rclcpp_components_register_node".

-- Configuring incomplete, errors occurred!
.....
```
- Operating System:
 ubuntu 18.04
- Installation type:
 ROS2 dashing(source code).

Is there any idea on solve this ? 
Thanks in advance.

clalancette, 2019-11-26:
It looks like you probably cloned the master version of the demos, which are currently targeting Foxy (and thus have features that Dashing doesn't have).  After cloning the `demos` repository, you need to switch to the `dashing` branch (`git checkout dashing`), then try building again.

Assuming that is the problem, I'm going to close this out.  Feel free to keep commenting on this issue.

closed 2019-11-26 by clalancette

friendsocket, 2019-11-27:
> It looks like you probably cloned the master version of the demos, which are currently targeting Foxy (and thus have features that Dashing doesn't have). After cloning the `demos` repository, you [...]

Yes, it is. thank you.