ros2/demos #432: image_tools cam2image  Is publishing a wrong frame_id

gonzodepedro, 2020-03-17:
## Bug report

**Required Info:**

- Operating System:
  -All
- Installation type:
  - Binaries
- Version or commit hash:
  - eloquent
- DDS implementation:
  - all

#### Steps to reproduce issue
Run cam2image demo

```
ros2 run image_tools cam2image
```

Check /image topic contents

```
ros2 topic echo /image
```

#### Expected behavior

Proper frame_id information is sent on the header of the Image message on the /image topic.

#### Actual behavior

frame_id is a number that increments for every message sent

```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: '401'
height: 240
width: 320
encoding: bgr8
is_bigendian: 0
step: 960
```

#### Additional information

This bug connects to https://github.com/ros2/rviz/issues/511

cardboardcode, 2020-07-06:
**TODO**:
Check if issue persists in ROS2 Foxy.

clalancette, 2021-06-10:
This was fixed by #433 a long time ago, so closing this out.

closed 2021-06-10 by clalancette