ros2/demos #438: Windows CI error: test_linktime_composition "Signal event received before subprocess transport available"

rotu, 2020-04-17:
The test_linktime_composition test seems to pass CI but it logs a runtime error "Signal event received before subprocess transport available" and the warning "Task exception was never retrieved"

https://ci.ros2.org/view/nightly/job/nightly_win_rel/1522/consoleText

```
test 8
      Start  8: test_linktime_composition__rmw_fastrtps_cpp

8: Test command: C:\Python37\python.exe "-u" "C:/ci/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/ci/ws/build/composition/test_results/composition/test_linktime_composition__rmw_fastrtps_cpp.xunit.xml" "--package-name" "composition" "--output-file" "C:/ci/ws/build/composition/launch_test/CHANGEME.txt" "--env" "RMW_IMPLEMENTATION=rmw_fastrtps_cpp" "--append-env" "AMENT_PREFIX_PATH=C:/ci/ws/build/composition/test_ament_index/Release" "PATH=C:/ci/ws/build/composition/Release" "--command" "C:/Python37/python.exe" "-m" "launch_testing.launch_test" "C:/ci/ws/build/composition/test_linktime_composition__rmw_fastrtps_cpp_Release.py" "--junit-xml=C:/ci/ws/build/composition/test_results/composition/test_linktime_composition__rmw_fastrtps_cpp.xunit.xml" "--package-name=composition"
8: Test timeout computed to be: 60
8: -- run_test.py: extra environment variables:
8:  - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
8: -- run_test.py: extra environment variables to append:
8:  - AMENT_PREFIX_PATH+=C:/ci/ws/build/composition/test_ament_index/Release
8:  - PATH+=C:/ci/ws/build/composition/Release
8: -- run_test.py: invoking following command in 'C:\ci\ws\build\composition':
8:  - C:/Python37/python.exe -m launch_testing.launch_test C:/ci/ws/build/composition/test_linktime_composition__rmw_fastrtps_cpp_Release.py --junit-xml=C:/ci/ws/build/composition/test_results/composition/test_linktime_composition__rmw_fastrtps_cpp.xunit.xml --package-name=composition
8: [INFO] [launch]: All log files can be found below C:\Users\ContainerAdministrator\.ros/log\2020-04-16-11-38-30-904499-48ec6e44c219-10832
8: [INFO] [launch]: Default logging verbosity is set to INFO
8: test_linktime_composition (composition.TestComposition)
8: Test process' output against expectations. ... skipped ''
8: 
8: ----------------------------------------------------------------------
8: Ran 1 test in 0.000s
8: 
8: OK (skipped=1)
8: Task exception was never retrieved
8: future: <Task finished coro=<LaunchService._process_one_event() done, defined at C:\ci\ws\install\Lib\site-packages\launch\launch_service.py:293> exception=RuntimeError('Signal event received before subprocess transport available.')>
8: Traceback (most recent call last):
8:   File "C:\ci\ws\install\Lib\site-packages\launch\launch_service.py", line 295, in _process_one_event
8:     await self.__process_event(next_event)
8:   File "C:\ci\ws\install\Lib\site-packages\launch\launch_service.py", line 315, in __process_event
8:     visit_all_entities_and_collect_futures(entity, self.__context))
8:   File "C:\ci\ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
8:     sub_entities = entity.visit(context)
8:   File "C:\ci\ws\install\Lib\site-packages\launch\action.py", line 108, in visit
8:     return self.execute(context)
8:   File "C:\ci\ws\install\Lib\site-packages\launch\actions\opaque_function.py", line 75, in execute
8:     return self.__function(context, *self.__args, **self.__kwargs)
8:   File "C:\ci\ws\install\Lib\site-packages\launch\actions\execute_process.py", line 421, in __on_signal_process_event
8:     raise RuntimeError('Signal event received before subprocess transport available.')
8: RuntimeError: Signal event received before subprocess transport available.
8: The link time registration of classes does not work on Windows
8: [INFO] [test_linktime_composition-1]: process started with pid [11612]
8: [test_linktime_composition-1] [INFO] [1587037110.961045100] [linktime_composition]: Load library 
8: [ERROR] [test_linktime_composition-1]: process[test_linktime_composition-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
8: [INFO] [test_linktime_composition-1]: sending signal 'SIGTERM' to process[test_linktime_composition-1]
8: [ERROR] [test_linktime_composition-1]: process has died [pid 11612, exit code 1, cmd 'C:/ci/ws/build/composition/Release/linktime_composition.exe'].
8: 
8: ----------------------------------------------------------------------
8: Ran 0 tests in 0.000s
8: 
8: OK
8: -- run_test.py: return code 0
8: -- run_test.py: verify result file 'C:/ci/ws/build/composition/test_results/composition/test_linktime_composition__rmw_fastrtps_cpp.xunit.xml'
 8/18 Test  #8: test_linktime_composition__rmw_fastrtps_cpp .....   Passed    5.72 sec
```

Blast545, 2020-05-12:
I think those runtime errors are caused by `colcon test` closing the test, not by the package itself. As you can see in the log, the test is being skipped (it can also be seen in [CI](https://ci.ros2.org/view/nightly/job/nightly_win_rel/1522/testReport/composition/TestComposition/)).

I was able to track the issue to the test file running this test, it is normal for test to be skipped as this feature is not supported on Windows. The exact line where the line is skipped is shown [here](https://github.com/ros2/demos/blob/1731d9def844af54d68186e43cfd0729e9a8610b/composition/test/test_linktime_composition.py.in#L48).

As `colcon test` is registering "OK (skipped=1)" and the message "The link time registration of classes does not work on Windows" I think this is working properly.

cottsay, 2020-05-12:
Pretty sure this is the same error I reported here: https://github.com/ros2/launch/issues/418

Blast545, 2020-05-12:
@cottsay Just checked your bug report and you are right, the problem it's the same. I think it should be tracked there.

rotu, 2020-05-12:
It may be the same issue. I don't understand `test_linktime_composition.py.in` enough to know whether it's using the launch framework correctly or not.

clalancette, 2024-07-26:
We don't see this anymore in e.g. https://ci.ros2.org/view/nightly/job/nightly_win_rel/3118 , so I'll close this out.

closed 2024-07-26 by clalancette