ros2/demos #449: Quality of Service "Liveliness" demo blocks

dabonnie, 2020-06-01:
## Bug report

**Required Info:**

- Operating System:
  - OSX 10.14.6
- Installation type:
  - [fat binary](https://ci.ros2.org/view/packaging/job/packaging_osx/1865/)
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - rclcpp and rclpy

#### Steps to reproduce issue
Run either of the commands

`ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000`

or

`ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000`

#### Expected behavior

Non-blocking behavior outputting liveliness events

#### Actual behavior

Blocking behavior and no liveliness events observed

#### Additional information

Found during testing [here]https://github.com/ros-tooling/aws-roadmap/issues/276#issuecomment-636101152). Note: this behavior was only observed for FastRTPS.

dabonnie, 2020-06-02:
Same behavior observed on Windows 10 (sever) using FastRTPS.

hidmic, 2020-06-02:
@dabonnie I'd think this is a duplicate of https://github.com/ros2/rmw_fastrtps/issues/384. Would you mind trying different policies?

sloretz, 2020-06-11:
@dabonnie friendly ping, does the issue @hidmic linked look like it's the cause of this one? If so, I'll close this one in favor of the earlier issue.

dabonnie, 2020-06-11:
> @dabonnie friendly ping, does the issue @hidmic linked look like it's the cause of this one? If so, I'll close this one in favor of the earlier issue.

Sorry, it does. Closing this in favor of https://github.com/ros2/rmw_fastrtps/issues/384.

closed 2020-06-11 by dabonnie